mavlink/MAVSDK #1600: When used the cancel of calibration plugin class, there is crash!!!

SSSnow, 2021-11-05:
debug the error:
A command cannot be sent if another command with the same command_id is still processing! Ignoring command 241
Mavsdk: Callback called from mavlink_commands.cpp:337 took more than 1 second to run.

when i comment the line:226 
https://github.com/mavlink/MAVSDK/blob/main/src/plugins/calibration/calibration_impl.cpp#:~:text=_parent-%3Esend_command_async(command%2C%20nullptr)%3B
"    _parent->send_command_async(command, nullptr);"

then work ok!   
I want to know why send a nullptr callback before send a normal cancel calibration command?
thanks!

SSSnow, 2021-11-05:
@bkueng

JonasVautherin, 2021-11-05:
Reading through the code, this seems like a bug indeed. I think this line should not be there.

@SSSnow: Would you mind opening a PR with that change?

JonasVautherin, 2021-11-12:
@SSSnow: Not sure I understood correctly: will you open a PR with your bugfix? You deserve the credit with the commit :blush:

PR #1986 (2023-02-10): calibration: remove spurious cancel command
This line must have slipped in there. It doesn't seem right.

Fixes #1600.
closed 2023-02-10 by julianoes