mavlink/MAVSDK #1603: Support cameras without camera definition file

dlech, 2021-11-06:
Currently, MAVSDK requires that cameras have a camera definition file, otherwise the camera plugin fails:

```
[08:23:45|Info ] Downloading camera definition from:  (camera_impl.cpp:1074)
[08:23:45|Error] Error while downloading text, curl error code: URL using bad/illegal format or missing URL (curl_wrapper.cpp:38)
[08:23:45|Error] Failed to download camera definition. (camera_impl.cpp:1076)
[08:23:45|Debug] Failed to fetch camera definition! (camera_impl.cpp:1030)
```

However, the [spec](https://mavlink.io/en/messages/common.html#CAMERA_INFORMATION) says:

> Camera definition URI (if any, otherwise only basic functions will be available).

So it seems like this should be a supported case.

JonasVautherin, 2021-11-06:
MAVSDK will try to download the camera definition file 3 times and then it will give up. Does that give you issues (like does it crash), or is it just that you don't want to see those error messages?

You're right that if `camera_information.cam_definition_uri` is an empty string, then [`bool CameraImpl::fetch_camera_definition`](https://github.com/mavlink/MAVSDK/blob/main/src/plugins/camera/camera_impl.cpp#L1102-L1103) should probably return false right away. Or something like that.

Would you mind trying it and opening a PR if that solves your problem? :blush:

dlech, 2021-11-06:
I guess I just didn't wait long enough. It would be nice to avoid the errors and the delay anyway. I'll have a look.

dlech, 2021-11-06:
I just checked again, and for the record, the camera prepare method does fail after the 3 retries.

PR #1606 (2021-11-08): camera: handle empty camera definition URI
Currently, MAVSDK expects all cameras to have a camera definition URI and the Prepare method of the camera plugin will fail after 3 tries to fetch the file.

However the spec says this is optional, so an empty string should be allowed and should avoid the error.

This adds a check for the empty string to `should_fetch_camera_definition()` to handle this case.

Fixes #1603
closed 2021-11-08 by julianoes