mavlink/MAVSDK #1610: battery information is not correct!!

imindude, 2021-11-08:
I'm using this in two case, one is MAVSDK/C++ and the other is MAVSDK-Server & C#.
Both of two cases, battery info. shows incorrect value. If I connect the 16V battery, it shows like this

In MAVSDK-Server & C# 

```
{ "voltageV": 15.1, "remainingPercent": 0.51 }
{ "voltageV": 15.1, "remainingPercent": 0.51 }
{ "voltageV": 3.775, "remainingPercent": 0.51 }
{ "voltageV": 3.775, "remainingPercent": 0.51 }
{ "voltageV": 3.775, "remainingPercent": 0.51 }
{ "voltageV": 3.775, "remainingPercent": 0.51 }
{ "voltageV": 3.775, "remainingPercent": 0.51 }
```

Beginning of some callbacks, it shows right value, but a few seconds later battery voltage is something wrong. Looks like one cell's voltage. In use of MAVSDK/C++, shows same result. So, my case - MAVSDK/C++ - I used the "mavlink_passthrough" plugin like this.

```
bool DroneModel_Impl::onPassthruIncomingMessageAsync(mavlink_message_t message)
{
  if (telemetry_armed_)
  {
    // logging here
  }

  if (message.msgid == MAVLINK_MSG_ID_RC_CHANNELS)
  {
    manual_control_ = (mavlink_msg_rc_channels_get_chan8_raw(&message) >= kForcedManualControlRcRaw);
  }
  else if (message.msgid == MAVLINK_MSG_ID_BATTERY_STATUS)
  {
    mavlink_battery_status_t  battery_status;
    float   voltage = 0.0f;

    mavlink_msg_battery_status_decode(&message, &battery_status);

    for (int i = 0; i < 10; i++)
    {
      if (battery_status.voltages[i] = UINT16_MAX)
        break;
      voltage += static_cast<float>(battery_status.voltages[i]) * 0.001f;
    }

    if (battery_status.id == 0)       tether_voltage_ = voltage;
    else if (battery_status.id == 1)  battery_voltage_ = voltage;
  }

  return true;
}
```

But MAVSDK-Server & C#, I think it can not access the raw mavlink message, isn't it?

Actually, I used the two power pack system. But same result that use only one power pack system. Telemetry plugin's Battery information is incorrect, my guess. Am I wrong? How can I get the battery voltage correctly?

julianoes, 2021-11-08:
Have you configured the battery parameters correctly?
https://docs.px4.io/master/en/advanced_config/parameter_reference.html#battery-calibration

imindude, 2021-11-08:
> Have you configured the battery parameters correctly? https://docs.px4.io/master/en/advanced_config/parameter_reference.html#battery-calibration

Yes. In QGroundControl shows correctly. Actually my code of above - onPassthruIncomingMessageAsync() is comes from QGroundControl's.

julianoes, 2021-11-08:
So the implemenation in MAVSDK is not correct? If so, can you suggest what is wrong, or make a pull request? Sorry, I don't have time to look into this in detail right now.

imindude, 2021-11-08:
> So the implemenation in MAVSDK is not correct? If so, can you suggest what is wrong, or make a pull request? Sorry, I don't have time to look into this in detail right now.

Here is QGC's battery code.

```
void VehicleBatteryFactGroup::_handleBatteryStatus(Vehicle* vehicle, mavlink_message_t& message)
{
    mavlink_battery_status_t batteryStatus;
    mavlink_msg_battery_status_decode(&message, &batteryStatus);

    VehicleBatteryFactGroup* group = _findOrAddBatteryGroupById(vehicle, batteryStatus.id);

    double totalVoltage = qQNaN();
    for (int i=0; i<10; i++) {
        double cellVoltage = batteryStatus.voltages[i] == UINT16_MAX ? qQNaN() : static_cast<double>(batteryStatus.voltages[i]) / 1000.0;
        if (qIsNaN(cellVoltage)) {
            break;
        }
        if (i == 0) {
            totalVoltage = cellVoltage;
        } else {
            totalVoltage += cellVoltage;
        }
    }

    group->function()->setRawValue          (batteryStatus.battery_function);
    group->type()->setRawValue              (batteryStatus.type);
    group->temperature()->setRawValue       (batteryStatus.temperature == INT16_MAX ?   qQNaN() : static_cast<double>(batteryStatus.temperature) / 100.0);
    group->voltage()->setRawValue           (totalVoltage);
    group->current()->setRawValue           (batteryStatus.current_battery == -1 ?      qQNaN() : static_cast<double>(batteryStatus.current_battery) / 100.0);
    group->mahConsumed()->setRawValue       (batteryStatus.current_consumed == -1  ?    qQNaN() : batteryStatus.current_consumed);
    group->percentRemaining()->setRawValue  (batteryStatus.battery_remaining == -1 ?    qQNaN() : batteryStatus.battery_remaining);
    group->timeRemaining()->setRawValue     (batteryStatus.time_remaining == 0 ?        qQNaN() : batteryStatus.time_remaining);
    group->chargeState()->setRawValue       (batteryStatus.charge_state);
    group->instantPower()->setRawValue      (totalVoltage * group->current()->rawValue().toDouble());
    group->_setTelemetryAvailable(true);
}
```

So, in the "MAVSDK/src/plugins/telemetry/telemetry_impl.cpp" line 1070 should be changed like this,

```
void TelemetryImpl::process_battery_status(const mavlink_message_t& message)
{
    mavlink_battery_status_t bat_status;
    mavlink_msg_battery_status_decode(&message, &bat_status);

    _has_bat_status = true;

    Telemetry::Battery new_battery;
    new_battery.id = bat_status.id;
#if 0
    new_battery.voltage_v = bat_status.voltages[0] * 1e-3f;
#else
    new_battery.voltage_v = 0.0f;
    for (int i = 0; i < 10; i++)
    {
        if (battery_status.voltages[i] == UINT16_MAX)
            break;
        new_battery.voltage_v += static_cast<float>(battery_status.voltages[i]) * 1e-3f;
    }
#endif
    // FIXME: it is strange calling it percent when the range goes from 0 to 1.
    new_battery.remaining_percent = bat_status.battery_remaining * 1e-2f;

    set_battery(new_battery);

    {
        std::lock_guard<std::mutex> lock(_subscription_mutex);
        if (_battery_subscription) {
            auto callback = _battery_subscription;
            auto arg = battery();
            _parent->call_user_callback([callback, arg]() { callback(arg); });
        }
    }
}
```

slaghuis, 2021-11-17:
I have implemented the fix as suggested by @imindude, and can confirm it works.  Battery voltage is now reporting correctly. p.s. I did a [git pull origin main] and recompiled.  I noticed the code is not in the release (v.0.47.0) that I got.

My other Raspberry Pi pulled v0.48.0, (no, I have no clue why, but I surely do not have the guts to update the file with these skills) and the code was not in there either.   Anyway, thank you for the fix.

JonasVautherin, 2021-11-17:
@imindude would you mind opening a PR with your proposed changes to MAVSDK? :blush:

JonasVautherin, 2021-11-17:
Resolved in https://github.com/mavlink/MAVSDK/pull/1621. Thanks!

closed 2021-11-17 by JonasVautherin

Amna-Smaoui, 2022-07-20:
Hi ! I'm using Mavsdk with python and I want get the battery.voltage and the battery.remaining_percent . At the begin it shows right value, but a few seconds later battery voltage is something wrong and i don't know what's wrong . I'm working with SITL not a ral drone .

julianoes, 2022-07-21:
@Amna-Smaoui please create a new issue and don't necrobump old ones.

Amna-Smaoui, 2022-07-21:
okey sorry ! I 'm gonna create new one

Amna-Smaoui, 2022-07-21:
I created a new issue . I hope you can help me figure out what's wrong
because I'm using it for my *graduation project* and I'm kinda  under *great
time pressure*.

Thank you for your understanding.
*Amna Smaoui *
*Élève ingénieur en génie électrique | ENIT*