mavlink/MAVSDK #1614: Heartbeats times out in SITL with multiple vehicles

marvin-wtt, 2021-11-09:
## Description

When running PX4 SITL simulation with gazebo and multiple plane models, MAVSDK keeps prompting an info output about timed out heartbeats and the discovery of the system shortly after.

| Application | Version |
|---|:-:|
| MAVSDK | 0.44.0 |
| PX4-Autopilot | 1.12.1 |
| gazebo | 11.8.1 |

As this behavior only happens with the plane model and only with multiple vehicles I'm not sure if this issue is MAVSDK or PX4 related but I thought it's best to start here :)

## Steps to reproduce

* Start PX4 gazebo sitl with at least two planes
`Tools/gazebo_sitl_multiple_run.sh -m plane -n 2`
* Start MAVSDK example
`./takeoff_and_land  udp://:14540`

## Output
```
[...]
[04:44:12|Info ] heartbeats timed out (system_impl.cpp:263)
[04:44:14|Debug] Discovered 1 component(s) (system_impl.cpp:520)
[...]
[04:44:17|Info ] heartbeats timed out (system_impl.cpp:263)
[04:44:19|Debug] Discovered 1 component(s) (system_impl.cpp:520)
[...]
```

julianoes, 2021-11-11:
Thanks for the issue. I won't have time to reproduce this in the next few days but I'll put it on my list.

JonasVautherin, 2021-11-12:
@marvin-wtt does it work with QGC? If not, it would suggest that maybe something is wrong with plane.

julianoes, 2021-12-01:
@marvin-wtt is this still happening? I would now have a chance to look into it.

marvin-wtt, 2021-12-01:
Yes, it is still happening but I noticed that the real time factor sometimes drops below 0.1 even though the CPU and GPU usage is very low. This probably causes the timeout. 
If you could confirm that this is the case at your end as well I would really appreciate it

julianoes, 2021-12-01:
Yes, that's likely causing this. You could increase the timeout for your build:
https://github.com/mavlink/MAVSDK/blob/21ac1ec4f4baabb278d774f6ec2d14a407041a64/src/core/system_impl.h#L372

closed 2021-12-01 by julianoes