mavlink/MAVSDK #1616: Missing Param - SITL. Not reading position

evanatherton1, 2021-11-09:
Trying to read global position using SITL. This is the output I get. Lat Lon and Alt values are read as nan.
[image]

julianoes, 2021-11-11:
What's your PX4 version and what's the MAVSDK version?

evanatherton1, 2021-11-12:
PX4: 1.12.3
I'm not sure how to check which version of MAVSDK is being run.

evanatherton1, 2021-11-12:
It looks like I'm running the latest version of mavsdk (0.48.0)

JonasVautherin, 2021-11-12:
Not sure I get the problem. What is the code you run, what do you expect as a result, and what do you get instead?

evanatherton1, 2021-11-15:
I'm just trying to print the location of the UAV (Latitude, Longitude, Altitude). After setting the update rate, I call my callback function as so:
`telemetry->subscribe_position(PositionCallbackFunc);`

`void PositionCallbackFunc(Telemetry::Position position)
{
    gDronePosition.alt = position.absolute_altitude_m;
    gDronePosition.lat = position.latitude_deg;
    gDronePosition.lon = position.longitude_deg;
    std::cout<< "Reported Drone Position: \r\n";
    std::cout << "Altitude: " << position.relative_altitude_m << " m" << std::endl
              << "Latitude: " << position.latitude_deg << std::endl
              << "Longitude: " << position.longitude_deg << std::endl;

 
    std::chrono::time_point<std::chrono::system_clock> now = std::chrono::system_clock::now();
    auto duration = now.time_since_epoch();
    m_drone_pos_update_Time = std::chrono::duration_cast<std::chrono::milliseconds>(duration).count();
}`

It never seems to enter the Callback function, and when I try to print the altitude latitude and longitude values they print as "nan"

JonasVautherin, 2021-11-15:
Did you try running an example, e.g. https://github.com/mavlink/MAVSDK/tree/main/examples/takeoff_and_land? Does that work?

evanatherton1, 2021-11-15:
[image]
I get this when attempting to build the example

JonasVautherin, 2021-11-15:
Did you install mavsdk on your system?

evanatherton1, 2021-11-15:
I believe so. I followed the instructions found here:
https://mavsdk.mavlink.io/v0.38/en/cpp/guide/build.html

JonasVautherin, 2021-11-15:
Your last screenshot shows a linking error when building the example, because the example tries to link against mavsdk. You have to solve that linking error, I cannot guess what's on your system :sweat_smile:

evanatherton1, 2021-11-15:
[image]
It looks like I'm having issues building the library. Would this be a place to start?

evanatherton1, 2021-11-15:
[image]
Looks like I'm having issues building the library. Would this be a place to start?

JonasVautherin, 2021-11-15:
If the cmake configure step fails ("Configuring incomplete, errors occurred"), then there is no need to try the build step :sweat_smile:.

It does not appear in your screenshot, but it seems like you are using the deprecated `-DBUILD_BACKEND=ON`, so probably you want to use a more recent version of the docs, e.g. just upstream: https://mavsdk.mavlink.io/main/en/cpp/guide/build.html.

Also to use MAVSDK-C++ (i.e. if you don't plan on using MAVSDK-Swift or MAVSDK-Java or MAVSDK-Python), no need to build mavsdk_server.

evanatherton1, 2021-11-16:
Followed the install instructions from that link. Specifically did this:

$ sudo apt-get update
$ sudo apt-get install build-essential cmake git
$ cd MAVSDK
$ git clone https://github.com/mavlink/MAVSDK.git
$ git submodule update --init --recursive
$ cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
$ cmake --build build/default -j8
$ cmake -Bbuild/default -DCMAKE_BUILD_TYPE=Release -H.
$ cmake --build build/default -j8
$ sudo cmake --build build/default --target install

Seemed to build and install properly. The linking error persists when I try to run the takeoff_and_land example. Specifically did this:
$ cd MAVSDK
$ cd examples
$ cd takeoff_and_land
$ cmake -Bbuild -H.
$ cmake --build build

Error messages come up - same as my 2nd most recent screenshot. Am I missing anything obvious?

JonasVautherin, 2021-11-16:
So `$ cmake -Bbuild -H.` in the example project worked and found mavsdk on the system?

evanatherton1, 2021-11-16:
When I run `$ cmake -Bbuild -H.` here's what I get:
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/MAVSDK/examples/takeoff_and_land/build

JonasVautherin, 2021-11-16:
And do you have the symbols in the build binaries? For me:

```
% nm ./build/default/src/core/libmavsdk.so | grep subscribe_on_new_system                                                       [0]
000000000017bd0e T _ZN6mavsdk10MavsdkImpl23subscribe_on_new_systemERKSt8functionIFvvEE
00000000001771d4 T _ZN6mavsdk6Mavsdk23subscribe_on_new_systemERKSt8functionIFvvEE
```

evanatherton1, 2021-11-17:
Where would I see that? Not sure how to access build binaries. 

Really appreciate the help. I'm new to this stuff

JonasVautherin, 2021-11-17:
Something like this should find it:

```
find ~/MAVSDK | grep libmavsdk.so
```

julianoes, 2024-05-27:
No more answer, closing.

closed 2024-05-27 by julianoes