mavlink/MAVSDK #1618: Can the gRPC server be started before mavsdk-server discovers/connects to a system?

divyanshu1234, 2021-11-16:
In some cases, applications using mavsdk may start before the UAV system starts and is discovered by the mavsdk-server. In this case, clients intending to connect to the mavsdk-server will effectively have to block because the gRPC port will not be available.

Alternatively, if the gRPC server is up and the port is available, the clients won't have to block and the channels that are running in the background will automatically start receiving data once the UAV system is discovered.

The earlier versions of the mavsdk-server used to behave in the same manner which made client-side implementation quite easy.

JonasVautherin, 2021-11-16:
Yeah it used to be that mavsdk would create a "dummy" system to receive subscriptions and whatnot, but that's not the case anymore. So your app has to wait until the drone is discovered before it can start the subscriptions. `mavsdk-server` is just reflecting that :innocent:

divyanshu1234, 2021-11-16:
I noticed this code in `mavsdk_server_api.cpp`:

```
    if (!mavsdk_server->connect(std::string(system_address))) {
        // Connection was cancelled
        return false;
    }

    auto grpc_port = mavsdk_server->startGrpcServer(mavsdk_server_port);
    if (grpc_port == 0) {
        // Server failed to start
        return false;
    }

    return true;
```

Are there any chances that putting `mavsdk_server->startGrpcServer(mavsdk_server_port)` before `mavsdk_server->connect(std::string(system_address))` would create any bugs?

Because in that case, the clients can push `mavsdk_server.run()` to the background thread and the gRPC port would be open for clients to connect before server blocks for systems.

JonasVautherin, 2021-11-16:
That won't work because the server implementation now needs to discover a drone. If you do that, your subscriptions won't ever have an effect because mavsdk won't save them for when a system is discovered.

Unfortunately I think that you have to deal with that on your side...

divyanshu1234, 2021-11-16:
Oh. I hadn't noticed when I was testing it because SITL was always there. Maybe I can make some changes to MAVSDK-Java so that the API works the same as before.

closed 2021-11-16 by divyanshu1234

JonasVautherin, 2021-11-16:
In my opinion, a very good next step for MAVSDK-Java would be to remove the dependency on gRPC and use direct bindings instead (i.e. JNI), so that it would be much more lightweight. But it also means that the API would get closer to the C++ API, which does not allow that at the moment.

And there you would see why this is problematic: MAVSDK-C++ supports discovering multiple drones, so if you subscribe to, say, the position updates before a drone is known, then MAVSDK cannot know if you were meaning to subscribe to the first or to the second drone that appears. That's why this is delegated to the user, because they know their use-case (and MAVSDK doesn't) :grin:.

divyanshu1234, 2021-11-16:
Yes indeed, and that would be an interesting project to take up. We not only use MAVSDK-Java for the Android GCS, but also on companion computers running our headless Java framework. And this would definitely give a performance boost to it.

But for now, I was thinking of altering some logic in the plugin wrappers that can give users a bit more flexibility without them having to worry about initialisation problems while keeping the API as simple as it was before. That certainly helped me a lot in adopting the library. And then we can look into JNI and multiple system support. I'll try to create a PR soon.