mavlink/MAVSDK #1625: Absence of mutex locking in Gimbal plugin causing SEGV_ACCERR

divyanshu1234, 2021-11-18:
While working on a new plugin mechanism for MAVSDK-Java I am getting the following error on Android: `Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR)`

This error comes only when the Gimbal plugin is enabled, and it signifies that two or more threads are mutating a variable without any locking. And I noticed that mutex locking is missing here:

https://github.com/mavlink/MAVSDK/blob/631f6d366a5683b773560d8abe9752b54afe98fa/src/plugins/gimbal/gimbal_impl.cpp#L68

I am not too well-versed in C++ and I am not sure if this is by design, but please look into this issue.

JonasVautherin, 2021-11-19:
This said, I'm tempted to remove this `_protocol_cookie` entirely, stay tune for a PR that you may test.

JonasVautherin, 2021-11-19:
This is the draft: https://github.com/mavlink/MAVSDK/pull/1627

divyanshu1234, 2021-11-19:
Great! I'll try to test it in a couple of days.

julianoes, 2024-05-27:
Should be fixed now.

closed 2024-05-27 by julianoes