mavlink/MAVSDK #1630: Allow more fine-grained offboard control

julianoes, 2021-11-22:
It has been requested - from what I remember - several times that it is possible to have better control around the auto-sending of offboard messages, and the way the mode is changed automatically.

Also see https://github.com/PX4/PX4-Autopilot/pull/18160 for more background.

This issue is to collect the requirements/wishlist.

benjinne, 2021-11-22:
I'm looking for a way to create PX4 MAVSDK tests to test offboard being lost. Currently, I only see the stop() function as a way to stop sending setpoints, however that also changes the flight mode which is problematic since it prevent's the offboard lost failsafe from happening. Also in the offboard failsafe tests, I would like to verify the correct flight mode was executed and not the `hold` mode selected by MAVSDK.

fas2037, 2021-11-26:
I need to send offboard set points to enable switching into offboard mode with the Remote Control. Currenty this is not really possible, as https://github.com/mavlink/MAVSDK/blob/7244ace0d7b223e8e7618d8736769f4f2a2ef1f3/src/plugins/offboard/offboard_impl.cpp#L757-L790 checks if the current flight mode is equal to Offboard and if not the sending of setpoints is canceled.

So I would suggest another function, similar to start, which enables the sending of setpoints without beeing in the Offboard flight mode.

JonasVautherin, 2021-11-26:
I don't get that. What's the use-case of sending offboard setpoints without being in offboard mode?

fas2037, 2021-11-26:
Being able to switch into offboard mode using the RC and not MAVSDK

marvin-wtt, 2021-11-26:
Switching into offboard mode (in this case via RC) is prohibited by PX4 until setpoints are received.
> The vehicle must be already be receiving a stream of target setpoints (>2Hz) before this mode - can be engaged.
MAVSDK however requires the vehicle to already be in offboard mode in order to start streaming setpoints. Thus, switching into offboard mode via RC is not possible.

MaEtUgR, 2021-11-30:
Just an idea: Since it's to specifically test an unplanned error case it might be useful to not build the functionality of the error case of losing offboard setpoints into the API but construct the test with artificial packet drops e.g. by using mavlink passthrough and then using the function intercept_outgoing_messages_async() to at some point drop the offboard setpoint messages. Maybe that's harder and less nice, not sure yet.

julianoes, 2021-11-30:
I think we could just add a config option where you can select whether you want automatic mode switch or not.

julianoes, 2021-11-30:
@MaEtUgR it's a good idea but it would mean I have to implement such a test, I assume.

fas2037, 2021-11-30:
> I think we could just add a config option where you can select whether you want automatic mode switch or not.

Sounds good, I think that would work fine for me.