mavlink/MAVSDK #1636: CPU load too high when downloading logs

Katawann, 2021-11-25:
Hello !

I download automatically the missing logs from the PX4 to my companion computer (RPi) thanks to the [LogFiles](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_log_files.html) plugin. The issue I have is when I'm downloading a log and I try to arm again. I get the error below. You can see this behavior on [this flight review](https://logs.px4.io/plot_app?log=8180f6b3-fbfb-4a3d-9173-5101d76bfe73).
`Warning: [PreFlightCheck] Fail: CPU load too high: 98.1%`

Is it possible to interrupt the [download_log_file_async](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_log_files.html#classmavsdk_1_1_log_files_1a48712c031e204b23e00cfa28a0b1bc01) function ? Otherwise is it possible to set a parameters to reduce the ressources used by the download to avoid this kind of situation ?

Thank you in advance for your suggestion !

julianoes, 2021-11-26:
I think that this is somewhat likely or expected. You could try to use Mavlink FTP as an alternative but it might have the same issue.

The logfile download implementation on PX4 is made in order to maximize the rate and serve the file as fast as possible, usually using the USB 1 interface. You could try to debug that by running `top` in the PX4 console and check just how much CPU is used by mavlink. From the log it looks like your base CPU usage is already at 80%, so then logfile download pushes it do 100%.

[image]

I don't have a good suggestion except not downloading logfiles while trying to fly, or dive in the PX4 code (or MAVSDK code?) to artificially slow it down.

Katawann, 2021-11-26:
Hi Julian,

thank you for the clear answer! Actually yes it makes sense that the download try to go as fast as possible. Therefore if I can not interrupt the download, I think the safest way is not to download anything when the drone is in use. 

I will put some energy to have some wake-up hardware/logic for the PX4 + companion computer in order to download the logs when the drone is in the garage of the customer or something like this

Thank you !

closed 2021-11-26 by Katawann