mavlink/MAVSDK #1637: Support multiple callback-subscribers

potaito, 2021-11-26:
MAVSDK allows to subscribe to new information with callbacks, like for example updates to the Position telemetry:

https://github.com/mavlink/MAVSDK/blob/7244ace0d7b223e8e7618d8736769f4f2a2ef1f3/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L990-L993

In the implementation only one callback function is registered, and subscribing a second time would simply remove the previous subscription silently:

https://github.com/mavlink/MAVSDK/blob/7244ace0d7b223e8e7618d8736769f4f2a2ef1f3/src/plugins/telemetry/telemetry_impl.cpp#L2075-L2079

I would love to have the possibility to add multiple callbacks / subscribers. This would also lower the chances, that I ruin someone else's subscriber in a collaborative code base by overwriting that person's callback function.

JonasVautherin, 2021-11-26:
Yep would be nice. Quite some work though, because all of the streams would need an `unsubscribe` function that would take a handle returned by the `subscribe` function, I would say :thinking:

potaito, 2021-11-26:
Sounds about right @JonasVautherin 

And yeah, the `unsubscribe` function will probably require some thoughts. Especially given that the same function could be added as callback multiple times, i.e. it's used in multiple places that don't know of each other.

I guess the solution will include an `std::vector` or `std::array` holding shared pointers to the callbacks. And then the pointer to the callback is only removed when the last usage of the shared point is gone.

JonasVautherin, 2021-11-26:
Yeah agreed. I think it would be worth looking into the cookie mechanism that @julianoes implemented for the timeouts. My first intuition is that such a "cookie" could be the handler that MAVSDK-C++ returns when you `auto position_handle = subscribe_position(some_callback)`.

When you call `unsubscribe_position(handle)`, then it unsubscribes the call back from that collection.

Now the annoying part is the language bindings: how does that handle look like in Swift, Python or Java? Maybe it can be just a type `Handle` that contains a `long` (the cookie)? Not sure.

julianoes, 2021-11-26:
Yes, something with a cookie/handle but nicer than what I did with `void *` that is super ugly, actually.
A `long` would already be better, yes :smile:. Weekend literature...

julianoes, 2021-11-29:
I started with a simple prototype for a registry: https://godbolt.org/z/hxhxsWc7M

Couple of points that are open:
- Probably needs to be thread-safe, or have locks around it.
- Should be one registry per mavsdk instance, or system, or per plugin, or per function?
- Should the callback itself be stored in a shared_ptr and then returned in the handle, so that if you destruct the handle, the callback is also destructed, and a weak_ptr in the registry can be checked and removed? It would be overhead but probably safer.

JonasVautherin, 2021-11-29:
1. I would definitely vote for thread-safe, otherwise we will always lock around it
2. I don't think that's necessary, I don't see why a user would want to destroy a handle :thinking: 
3. I'd say per system. Definitely not in `Mavsdk`, as `System` should be as independent from it as possible. Per plugin/function sound like they will have their own edge cases (what if 2 functions from 2 different plugins want to subscribe to the same Event?)
4. What about a `Map<Event, vector<Entry>>`? What's called often is the `call(Event event)` function, and it applies only to the list of callbacks that are interested in that event.

julianoes, 2022-06-02:
FYI: this is coming soon in version 2: https://github.com/mavlink/MAVSDK/pull/1768.