mavlink/MAVSDK #1645: question: how can I get the RC inputs from mavsdk_server ?

imindude, 2021-12-03:
I'm making some autocontrol in use of the mavsdk_server and C#.
I think almost feature is exist that I need, except one.
In emergency situation, I want to stop the drone by kill switch that is controlled by RC.
Mavsdk, it support the mavlink_passthrough feature. So I was using this and implemented emergency stop feature.
But I think mavsdk_server did not support the passthrough or callback of RC input.
How can I know the RC input in C# application? mavsdk_server can handle the RC input?

julianoes, 2021-12-03:
PX4 should switch out of an auto mode automatically if you move the sticks in RC.

imindude, 2021-12-03:
> PX4 should switch out of an auto mode automatically if you move the sticks in RC.

Yes, I know. Because of this, for safety, I am trying to check and keep the offboard mode continuously. Because pilot can move the RC sticks unexpectedly. In offboard mode, something induce some problems and computer is not know, the pilot should intercept the control authority(?). I mean computer worked correctly but drone's flight seems bad. In this case, pilot should be stop the offboard mode in use of RC and take the drone's control. But as I told above, computer is keeping the offboard continuously prevent pilot's mis-control.

In this case, if computer can know the RC input, pilot let computer know to stop the offboard mode clearly. So, I'm implement this feature with MAVSDK (mavlink-passthrough plugin). If mavsdk-server can do this, more helpful, my guess. Like a C#+UWP, java or something.

julianoes, 2021-12-03:
Aha, so you are saying offboard mode from MAVSDK keeps switching back to offboard? Can you share a log where that happens? I think that would not be intentional.

imindude, 2021-12-03:
> Aha, so you are saying offboard mode from MAVSDK keeps switching back to offboard? Can you share a log where that happens? I think that would not be intentional.

Sorry for my english. I want to select the offboard mode or not by RC switch (not gimbal stick). In default state of MAVSDK, if someone touch the RC stick, it exit the offboard mode. I do not want this. So, my control program check the flight mode always. If not offboard mode, it try to change the offboard mode immediately.

But drone shows instable flight, I want to change the manual control by RC switch (not stick). So I implemented this feature - offboard mode or not by RC switch - in use of MAVSDK's mavlink-passthrough plugin like this.

```
bool DroneModel_Impl::onPassthruIncomingMessageAsync(mavlink_message_t message)
{
  if (telemetry_armed_)
  {
    // logging here
  }

  if (message.msgid == MAVLINK_MSG_ID_RC_CHANNELS)
  {
    manual_control_ = (mavlink_msg_rc_channels_get_chan8_raw(&message) >= kForcedManualControlRcRaw);
  }
  else if (message.msgid == MAVLINK_MSG_ID_BATTERY_STATUS)
  {
    mavlink_battery_status_t  battery_status;
    float   voltage = 0.0f;

    mavlink_msg_battery_status_decode(&message, &battery_status);

    for (int i = 0; i < 10; i++)
    {
      if (battery_status.voltages[i] == UINT16_MAX)
        break;
      voltage += static_cast<float>(battery_status.voltages[i]) * 0.001f;
    }

    if (battery_status.id == 0)       tether_voltage_ = voltage;
    else if (battery_status.id == 1)  battery_voltage_ = voltage;
  }

  return true;
}
```

As you can see, if RC channel 8's raw data show high value, my control program change the flight mode to manual immediately.

These days, I check the MAVSDK SERVER because sometime need C# or java something. In this case MAVSDK is not acceptable, my guess. But MAVSDK SERVER does not have this feature - mavlink passthrough or read the RC channel value.

Am I wrong? Can I read the RC input value in case of MAVSDK SERVER ?

julianoes, 2021-12-03:
Oh, I think you understand. So instead of detecting it in MAVSDK, you should directly check that switch in PX4, either using [RC_MAP_OFFB_SW](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#RC_MAP_OFFB_SW), or using the [RC_MAP_FLTMODE](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#RC_MAP_FLTMODE). You can configure these in QGC.

Also, do enable or disable override from the sticks, there is a param that you can use: [COM_RC_OVERRIDE](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#COM_RC_OVERRIDE)

imindude, 2021-12-03:
Aha. Thank you. I'll check that.

closed 2024-05-27 by julianoes