mavlink/MAVSDK #1656: C++ No Mission Progress

tareed, 2021-12-28:
Latest MAVSDK: 0.50.1
Latest PX4 Simualtor (jMavsim): 1.12.3
Latest QGroundControl: 4.2.0

Trying to get the current mission item by subscribing to the mission progress, we do not get the correct position or updates.  Sometimes on startup the position will be returned at -1, when the sim gets to the first waypoint no update is received.  Attached code is a sample of the app that we are running and demonstrates the issue.  We are using this with no issues in the Python library.

A side note, the mission plan was created in QGC and uploaded to the sim. When the C++ attempts to download the plan, an error is thrown by MAVSDK.  "UNSUPPORTED mission item command (22)"  Not sure if that has something to do with what I am seeing.  What in the plan needs to be changed to remove this error?  QGC was configured to be used against PX4, not ArduPilot.

`#include <iostream>
#include <cstdint>
#include <thread>
#include <chrono>
#include <sstream>
#include <iomanip>
#include <chrono>
#include <math.h>
#include <csignal>
#include <string>

#include <future>

#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/telemetry/telemetry.h>
#include <mavsdk/plugins/mission/mission.h>
#include <mavsdk/plugins/action/action.h>

static void telemetryStatus(std::shared_ptr<mavsdk::System> system, const int port);

void signalHandler(int signum) {
	std::cout << "Interrupt signal (" << signum << ") received" << std::endl;

	std::cout << "---- EXITING ----" << std::endl;
	exit(signum);
}

int main(int argc, char* argv[]) {
	try {
		if(argc < 2 ) {
			std::cerr << "USAGE: orb <connection URL> <MAV ID>" << std::endl;
			return 1;
		}

		// Register the signal handler
		signal(SIGINT, signalHandler);

		mavsdk::Mavsdk mavsdk;

		int argCtr = argc-1;
		int mavid = std::stoi(argv[argCtr]);

		// Read the connection URL parameters
		for(size_t i = 1; i < (argCtr); ++i) {
			std::cout << "Setting up listener on: " << argv[i] << std::endl;
			mavsdk::ConnectionResult connResult = mavsdk.add_any_connection(argv[i]);

			if(connResult != mavsdk::ConnectionResult::Success) {
				std::cerr << "Connection Error: " << connResult << std::endl;
				return -1;
			}
		}

		std::atomic<size_t> numSystemsDiscovered{0};

		std::cout << "Waiting to discovery systems..." << std::endl;
		mavsdk.subscribe_on_new_system([&mavsdk, &numSystemsDiscovered]() {
			const auto systems = mavsdk.systems();

			if(systems.size() > numSystemsDiscovered) {
				std::cout << "System Discovered" << std::endl;
				numSystemsDiscovered = systems.size();
			}
		});

		std::this_thread::sleep_for(std::chrono::seconds(2));

		std::vector<std::shared_ptr<std::thread>> threads;

		for (auto system : mavsdk.systems()) {
			const int systemID = (*system.get()).get_system_id();
			std::cout << "System ID:" << systemID << " MAVID: " << mavid << std::endl;

			//Check to see if this is the aircraft we are supposed to listen for
			if(systemID == mavid) {
				std::cout << "Starting Thread For System ID: " << systemID << std::endl;
				int port = 20200 + mavid;

				threads.push_back(std::make_shared<std::thread>(&telemetryStatus, std::ref(system), port));
				std::this_thread::sleep_for(std::chrono::seconds(1));

				// TODO
				auto mission = mavsdk::Mission{system};
				std::atomic<bool> want_to_pause{false};
				std::cout << "Setting up Mission Listener" << std::endl;

				mission.subscribe_mission_progress([&want_to_pause](mavsdk::Mission::MissionProgress mission_progress) {
					std::cout << "--->Current Mission Item: " << mission_progress.current << std::endl;
				});

                auto action = mavsdk::Action{system};
                std::cout << "Arming..." << std::endl;
                const mavsdk::Action::Result arm_result = action.arm();
                if (arm_result != mavsdk::Action::Result::Success) {
                    std::cerr << "Arming failed: " << arm_result;
                    return 1;
                }
                std::cout  << "Armed." << std::endl;
                std::pair<mavsdk::Mission::Result, mavsdk::Mission::MissionPlan> plan = mission.download_mission();
                
                mavsdk::Mission::Result startMissionResult = mission.start_mission();
                if(startMissionResult != mavsdk::Mission::Result::Success) {
                    std::cerr << "Unable to start mission!" << std::endl;
                    return 1;
                }
			}
		}

		std::cout << "About to join threads" << std::endl;
		for (auto& t : threads) {
			std::cout << "Joining Threads" << std::endl;
			t->join();
		}

	} catch(const std::exception &ex) {
		std::cerr << "ERROR : " << ex.what() << std::endl;
		return -1;
	}

	return 0;
}

void telemetryStatus(std::shared_ptr<mavsdk::System> system, const int port) {

	try {
		auto telemetry = mavsdk::Telemetry{system};
		const int systemID = (*system.get()).get_system_id();

		std::cout << "Thread started for System ID: " << systemID << std::endl;

		//Check to see if we are looking at an invalid ID
		if (systemID == 255) {
			return;
		}

		// We want to listen to the position of the drone at 1 Hz.
		const mavsdk::Telemetry::Result set_rate_result = telemetry.set_rate_position(1.0);

		if (set_rate_result != mavsdk::Telemetry::Result::Success) {
			std::cerr << "ERROR: Setting rate failed:" << set_rate_result << std::endl;
			//return;
		}

		while(1) {
			try {
				mavsdk::Telemetry::Position position = telemetry.position();
				mavsdk::Telemetry::VelocityNed velocityNed = telemetry.velocity_ned();

				//Check to make sure the location is valid
				if(!isnan(position.latitude_deg) && !isnan(position.longitude_deg)) {
                    //Do WORK
                    std::cout << "Telem: " << position.latitude_deg << ":" << position.longitude_deg << std::endl; 
				}

				std::this_thread::sleep_for(std::chrono::milliseconds(1000));
			} catch(const std::exception &ex) {
				std::cerr << "ERROR processing telemetry data: " << ex.what() << std::endl;
			}
		}

	} catch(const std::exception &ex) {
		std::cerr << "ERROR : " << ex.what() << std::endl;
	}

	return;
}
`

tareed, 2021-12-29:
OBE

closed 2021-12-29 by tareed

JonasVautherin, 2021-12-29:
OBE? What's that?

tareed, 2021-12-29:
“Overcome By Events”.  I submitted this just before I received an answer on Slack.  It can really be deleted.