mavlink/MAVSDK #1665: MAVSDK collision avoidance

PeymanAmirii, 2022-01-11:
Hi,
Does MAVSDK have obstacle avoidance feature? If yes, how we can activate it?
If no, can we use MAVROS besides MAVSDK to activate this feature?
Many Thanks.

julianoes, 2022-01-11:
It does not in itself. MAVSDK is just an interface to vehicle supporting mavlink, so it depends on what the drone can do.

What exactly are you trying to do?

Also have a look at https://github.com/PX4/PX4-Avoidance.

PeymanAmirii, 2022-01-11:
Thanks for your quick reply. On https://github.com/PX4/PX4-Avoidance , it is mentioned that "X4 computer vision algorithms packaged as ROS nodes for depth sensor fusion and obstacle avoidance". So avoidance feature just can be used by MAVROS not MAVSDK, am I right?

julianoes, 2022-01-11:
I would recommend to use MAVROS for that purpose, yes.

AzeemMuzammil, 2022-01-13:
Hi, I have used, the above ROS implementation of obstacle avoidance (https://github.com/PX4/PX4-Avoidance) and when I plan the path using QGroundControl it works perfectly. Now I have a code written in Python using MAVSDK to control drone in offboard.

when I start a gazebo simulation through the command `make px4_sitl gazebo` and if I run the code it works perfectly. but when I run the drone through ROS in gazebo, my code doesn't connect to the drone. Can you check the issue please?

Part of my code that connects to the drone is 

```
from mavsdk import System

async def run():
        drone = System()
        await drone.connect(system_address="udp://:14540")
```

julianoes, 2022-01-13:
MAVROS also uses the port 14540, so that's probably why you can't have both connected at the same time.

You can also use port 14550 but that's usually used by QGC.

If you need more ports, you should be able to add one like this:

https://github.com/PX4/PX4-Autopilot/blob/8a01135a937db05ad812bbf412adff74562e8486/ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink#L26

AzeemMuzammil, 2022-01-13:
Thanks @julianoes

BTW, `mavlink start -x -u $udp_offboard_port_local -r 4000000 -f -m onboard -o $udp_offboard_port_remote` in  this command, what is offboard_port_local and remote? is it 14540 the local? and should I add a random unused port for remote? and which port should I use to connect to the drone? remote or local?

julianoes, 2022-01-13:
See here:
https://github.com/PX4/PX4-Autopilot/blob/8a01135a937db05ad812bbf412adff74562e8486/ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink#L5

`udp_offboard_port_remote` is 14540, the one that MAVSDK requires. The other one can be any free port.

AzeemMuzammil, 2022-01-13:
Hi,
I ran something like this in px4 shell, `pxh> mavlink start -x -u 14570 -r 4000000 -f -m onboard -o 14540` and it printed something like this `INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14570 remote port 14540`

But, even after this, if I rewrite the code to this, it doesent work. I mean doesent connects to the drone

```
from mavsdk import System

async def run():
        drone = System()
        await drone.connect(system_address="udp://:14570")
```

PeymanAmirii, 2022-01-13:
Hi Azeem,
Did you use offboard of MAVSDK along with Collision avoidance of ROS? Did it work correctly?

AzeemMuzammil, 2022-01-13:
Hi @PeymanAmirii 
Offboard mode doesn't avoid obstacle, but when I run in mission(auto) mode. it worked. Do you have any idea on making it work for offboard mode as well?

dmeirap, 2022-01-13:
Hii!! I'm also trying to avoid obstacles using MAVSDK. Have you figured out how to add the port correctly to use mavros and mavlink??

julianoes, 2022-01-13:
@AzeemMuzammil that won't work. You're using 14540 in your line which is already used.

Try something like this:

```
mavlink start -x -u 14570 -r 4000000 -f -m onboard -o 17873
```

AzeemMuzammil, 2022-01-13:
@julianoes thanks and, it works :). BTW do you have any idea why obstacle avoidance doesn't work with off-board mode?

AzeemMuzammil, 2022-02-03:
Hi @julianoes

When I run the command `make px4_sitl gazebo` I get the px4 console where I can enter this command `mavlink start -x -u 14570 -r 4000000 -f -m onboard -o 17873` to add a new port which I use for my separate program to control the drone.

But, in a multi-drone simulation scenario, when I launch like 3 drones with this command `Tools/gazebo_sitl_multiple_run.sh -n 3 -w baylands`, I have no access to the px4 console to run the command to enable additional ports. In this case, how can I enable additional ports? when I don't have access to the px4 console. Is there any way I can attach a px4 console to each drone in the simulation?

julianoes, 2022-02-04:
You need to add it to the the startup script here:
https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink

TSC21, 2022-02-11:
> @julianoes thanks and, it works :). BTW do you have any idea why obstacle avoidance doesn't work with off-board mode?

Because Avoidance is only aimed to function in Auto modes in PX4.

julianoes, 2022-03-18:
> When I run the same script from the remote machine with the system_address being set to:
> await drone.connect(system_address="udp://:14641")
> nothing happens. Could you please clarify this?

`udp://:14641` will bind to the port locally, so the drone on the remote system needs to talk to the "local" IP to reach MAVSDK.

To do that you need to tell the remote one where to send the traffic like so:
- https://docs.px4.io/master/en/simulation/#enable-udp-broadcasting, or:
- https://docs.px4.io/master/en/simulation/#enable-streaming-to-specific-address

tpaluavio, 2022-11-22:
Hi, I trying to perform collision prevention with px4 offboard mode with mavros/mavsdk. Can anyone point me to a spacefic link for that?