mavlink/MAVSDK #1673: package architecture (amd64) does not match system (arm64)

CY-1992, 2022-01-18:
Hello,

I am trying to install mavsdk library package (without mavsdk server) on linux arm64. DPKG could not process the package as the architecture is different. To reproduce the error, the following docker file can be used.

```
FROM arm64v8/ubuntu:20.04

RUN apt-get -y clean --fix-missing && \
    apt-get -y update --fix-missing && \
    DEBIAN_FRONTEND=noninteractive apt-get install -y build-essential cmake git wget && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean

WORKDIR /home/dep
RUN wget https://github.com/mavlink/MAVSDK/releases/download/v1.0.1/mavsdk_1.0.1_debian11_arm64.deb && \
    dpkg -i mavsdk_1.0.1_debian11_arm64.deb
```
Building docker image using buildx
```
sudo docker buildx build --platform=linux/arm64 .
```
Error logs
```bash

#7 21.09 Get:15 http://ports.ubuntu.com/ubuntu-ports focal-security/restricted arm64 Packages [3300 B]
#7 21.09 Get:16 http://ports.ubuntu.com/ubuntu-ports focal-security/multiverse arm64 Packages [3232 B]
#7 21.09 Get:17 http://ports.ubuntu.com/ubuntu-ports focal-security/main arm64 Packages [931 kB]
#7 21.62 Get:18 http://ports.ubuntu.com/ubuntu-ports focal-security/universe arm64 Packages [775 kB]
#7 22.53 Fetched 17.3 MB in 14s (1226 kB/s)
#7 22.53 Reading package lists...
#7 25.96 dpkg: error processing archive mavsdk_1.0.1_debian11_arm64.deb (--install):
#7 25.96  package architecture (amd64) does not match system (arm64)
#7 25.99 Errors were encountered while processing:
#7 25.99  mavsdk_1.0.1_debian11_arm64.deb
------
error: failed to solve: executor failed running [/bin/sh -c wget https://github.com/mavlink/MAVSDK/releases/download/v1.0.1/mavsdk_1.0.1_debian11_arm64.deb &&     dpkg --add-architecture arm64 &&     apt-get -y update --fix-missing &&     dpkg -i mavsdk_1.0.1_debian11_arm64.deb]: exit code: 1
```
We can build mavsdk from the source using cmake. But using package would speed up our build process.

julianoes, 2022-01-18:
Thanks for testing. Sounds like we need to fix that package!

julianoes, 2022-01-18:
Let's see if #1674 fixes it.

julianoes, 2022-01-18:
Let's see if https://github.com/mavlink/MAVSDK/releases/tag/v1.0.2 works for you.

CY-1992, 2022-01-18:
Thanks @julianoes  . It is installed successfully. But I encountered one more problem while building the example takeoff_and_land.cpp
`cmake -Bbuild  -H.`

Error logs
```bash
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/lib/cmake/MAVSDK/MAVSDKConfig.cmake:7 (find_dependency)
  CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/dep/build/CMakeFiles/CMakeOutput.log".

```

JonasVautherin, 2022-01-18:
The MAVSDK package does not come with curl, does it? Is curl installed on your system?

julianoes, 2022-01-19:
@JonasVautherin I was assuming everything would be included because I was building a static library with superbuild, but I guess that's not what happened.

JonasVautherin, 2022-01-19:
Hmm the MAVSDKConfig.cmake we provide does not do anything for static builds, so it will still try to link the dependencies. The static build works for `mavsdk_server`, as an executable, but it was not meant to be used for packaging in a distro. Not sure if we should do that to be honest, I would rather have the packages depend on the system libraries (I missed that, I did not realize it was a static build).

What do you think?

CY-1992, 2022-01-19:
I installed [amd64.deb](https://github.com/mavlink/MAVSDK/releases/download/v1.0.2/mavsdk_1.0.2_ubuntu20.04_amd64.deb) on ubuntu amd64. It worked perfectly fine without installation of any dependencies. So I assumed that dependencies would be included in [arm64.deb](https://github.com/mavlink/MAVSDK/releases/download/v1.0.2/mavsdk_1.0.2_debian11_arm64.deb).

julianoes, 2022-01-19:
I can reproduce it on a RPi 4. And I installed `libcurl4-openssl-dev` and it's still the same problem, and I don't actually see the curl headers on the system.

julianoes, 2022-01-19:
@JonasVautherin you're right. let's make it a dynamic build and require the dependencies.

julianoes, 2022-01-19:
https://github.com/mavlink/MAVSDK/pull/1675

julianoes, 2022-01-19:
Ok, the release v1.0.3 now configures successfully, but it fails to link for me on Ubuntu 20.04 because the glibc version is too old there
```
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `stat64@GLIBC_2.33'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `std::__atomic_futex_unsigned_base::_M_futex_wait_until_steady(unsigned int*, unsigned int, bool, std::chrono::duration<long, std::ratio<1l, 1l> >, std::chrono::duration<long, std::ratio<1l, 1000000000l> >)@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `pthread_detach@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `__libc_single_threaded@GLIBC_2.32'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `pthread_join@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `stat@GLIBC_2.33'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `pthread_create@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.3: undefined reference to `pthread_cond_clockwait@GLIBC_2.34'
```

CY-1992, 2022-01-19:
Same error with release v1.0.4

```bash
Scanning dependencies of target takeoff_and_land
[ 50%] Building CXX object CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o
[100%] Linking CXX executable takeoff_and_land
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `stat64@GLIBC_2.33'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `std::__atomic_futex_unsigned_base::_M_futex_wait_until_steady(unsigned int*, unsigned int, bool, std::chrono::duration<long, std::ratio<1l, 1l> >, std::chrono::duration<long, std::ratio<1l, 1000000000l> >)@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `pthread_detach@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `__libc_single_threaded@GLIBC_2.32'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `pthread_join@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `stat@GLIBC_2.33'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `pthread_create@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.4: undefined reference to `pthread_cond_clockwait@GLIBC_2.34'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/takeoff_and_land.dir/build.make:85: takeoff_and_land] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/takeoff_and_land.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```

julianoes, 2022-01-19:
Yep, working on it. I believe I needed to clear the cache.

julianoes, 2022-01-19:
I got the docker tags wrong, fix is coming.

julianoes, 2022-01-20:
Still wrong, working on it.

julianoes, 2022-01-20:
This is now working with v1.0.8 on aarch64 on a RPi 4 with Debian Buster, as well as armv6 on a RPi 3.

closed 2022-01-20 by julianoes

CY-1992, 2022-01-20:
Thanks @julianoes . It is working :smile: .

PR #1674 (2022-01-18): tools/workflows: fix packages for arm
Without the arch argument, fpm seems to use amd64 instead which is wrong for the arch packages.

Hopefully resolves #1673.