mavlink/MAVSDK #1679: MAVSDK Windows Installation

Tzhao7, 2022-01-19:
Hi, 
When installing MAVSDK on Windows, are we given the option to either 1) locally install mavsdk-windows-x64-release.zip or 2) clone the repository and cmake build and install? The zip folder does not contain a CMakeLists.txt so do we have to make one ourselves? Because of this uncertainty I tried the second approach by using:
git clone https://github.com/mavlink/MAVSDK.git
git submodule update --init --recursive
cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -S. 
cmake --build build/default -j8 
And was given this error message on the last command:
[image]
Note the message is a lot longer but the errors all involve unresolved external symbols.

Tzhao7, 2022-01-19:
I was just looking through previous Windows installation issues and came across this: https://github.com/mavlink/MAVSDK/issues/1357#issue-832349620. Used the same commands as the user and think I've managed to build and install without errors. Could you confirm that these are the correct commands for the debug version?

julianoes, 2022-01-20:
Your compiling commands look correct. I wonder why the linking fails for you but not in CI? Which MSVC version are you using? I'll try to reproduce this in my Windows later.

Tzhao7, 2022-01-20:
I'm using "MSVC 19.29.30136.0".

julianoes, 2022-01-21:
I see the same on my local machine:

```
`telemetry_health.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl(enum
mavsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,clas
s std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDetailed::
callback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_traits@D@s
td@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runner.vcx
proj]
telemetry_sync.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl(enum ma
vsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class
std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDetailed::ca
llback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_traits@D@std
@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runner.vcxpr
oj]
mavlink_passthrough.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl(en
um mavsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,c
lass std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDetaile
d::callback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_traits@
D@std@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runner.
vcxproj]
mission_transfer_lossy.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl
(enum mavsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const
&,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDeta
iled::callback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_trai
ts@D@std@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runn
er.vcxproj]
system_connection_async.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdec
l(enum mavsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const
 &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDet
ailed::callback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_tra
its@D@std@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_run
ner.vcxproj]
statustext.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl(enum mavsdk
::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std:
:basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDetailed::callba
ck" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_traits@D@std@@V?
$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runner.vcxproj]
telemetry_async.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl(enum m
avsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class
 std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDetailed::c
allback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_traits@D@st
d@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runner.vcxp
roj]
telemetry_gps_origin.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl(e
num mavsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,
class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDetail
ed::callback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_traits
@D@std@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runner
.vcxproj]
offboard_acceleration.obj : error LNK2001: unresolved external symbol "public: static class std::function<bool __cdecl(
enum mavsdk::log::Level,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &
,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)> mavsdk::LogDetai
led::callback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_string@DU?$char_trait
s@D@std@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runne
[... cut ...]
or<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)
> mavsdk::LogDetailed::callback" (?callback@LogDetailed@mavsdk@@2V?$function@$$A6A_NW4Level@log@mavsdk@@AEBV?$basic_str
ing@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@1H@Z@std@@A) [C:\src\MAVSDK\build\default\src\integration_tests\integ
ration_tests_runner.vcxproj]
C:\src\MAVSDK\build\default\src\integration_tests\Debug\integration_tests_runner.exe : fatal error LNK1120: 1 unresolve
d externals [C:\src\MAVSDK\build\default\src\integration_tests\integration_tests_runner.vcxproj]
```

PR #1686 (2022-01-21): core: fix linking on Windows
This might fix #1679.
closed 2022-01-21 by julianoes