mavlink/MAVSDK #1683: Handling of MAVLINK params

ykhedar, 2022-01-20:
I just stumbled upon the function which handles the MAV_PARAM types and found that the function in :

https://github.com/mavlink/MAVSDK/blob/a6836be251e216f9967183a7551cc6ccb2a60382/src/mavsdk/core/mavlink_parameters.cpp#L923

handles it in a different way then the function for handling MAV_PARAM_EXT types. Is there a reason for this? Even the setting of `_value` in this function seems to be odd. Since the `temp.int32_value` is never assigned a value, using it to assign it to `_value` seems to give it garbage values. Or am I missing something?

julianoes, 2022-01-20:
Can you describe the problem that you're trying to solve? I'm not sure I understand what you mean.

The `int32_value` is a union with the `float_value`, so it's accessing the same bits but with case to a different type.

julianoes, 2022-01-20:
@bazfp no, why? So you're saying we should use `std::variant` instead? I mean we should anyway but I don't understand why it's UB.

ykhedar, 2022-01-21:
> Can you describe the problem that you're trying to solve? I'm not sure I understand what you mean.
> 
> The `int32_value` is a union with the `float_value`, so it's accessing the same bits but with case to a different type.

I was exploring why getting all parameters from ardupilot sitl using MAVSDK was failing for some type of parameters. Since PX4 has only int32 and float param types (apm has in addition int8, int16 as well), the int8 and int16 param type was not being handled in the function tagged in my OP. 

Doing more investigations, I will post in more detail once the tests are done completely.

julianoes, 2022-01-21:
> apm has in addition int8, int16 as well), the int8 and int16 param type was not being handled in the function tagged in my OP.

Aha, I understand! I was under the impression ArduPilot only has float and nothing else.

ykhedar, 2022-02-07:
Hi again, 

In the past few days I did some more research on the parameter handling related to PX4, Ardupilot and the current state in MAVSDK.

MAVLINK standard supports: uint8, int8, uint16, int16, uint32, int32, uint64, int64, real32, real64
PX4 supports/uses:                int32 and real32(float)
as found from [PX4 Param Reference](https://docs.px4.io/v1.9.0/en/advanced_config/parameter_reference.html)

Ardupilot supports/uses:         int8, int16, int32, real32(float)
as found from [APM Copter Param Reference](https://ardupilot.org/dev/docs/mavlink-get-set-params.html#setting-a-parameter)

MAVSDK supports int32 and float(real32)

Desired behavior for a Autopilot Param service:
1. getting a param
2. setting a param
3. getting all params

The desired behavior works for PX4 autopilot. Now to support the other data types in the param service, i have considered the following two options:

**Option A: Change the param.proto file to also support getting and setting of other data types.** 

This is not possible sadly since protobuf do not support int8, int16. Dont know if there is a workaround this.

**Option B: Making changes to mavlink_parameters.cpp.**

The following function: 
https://github.com/mavlink/MAVSDK/blob/9b4b79cbdfca361fab9a528ba78e81376503cbd0/src/mavsdk/core/mavlink_parameters.cpp#L455

handles the parameters exchanged between the mavsdk and autopilot. 

In this function, the block of code in the if statement (which is to process the response of GetAllParams) works for ardupilot as well if I make the necessary changes in the function:

https://github.com/mavlink/MAVSDK/blob/9b4b79cbdfca361fab9a528ba78e81376503cbd0/src/mavsdk/core/mavlink_parameters.cpp#L915

The changes are mostly to add cases for missing data types (int8 and int16 for ardupilot).

The problem comes while GetParamInt/GetParamFloat since then the function to check the data type of the param is called:

https://github.com/mavlink/MAVSDK/blob/9b4b79cbdfca361fab9a528ba78e81376503cbd0/src/mavsdk/core/mavlink_parameters.cpp#L508

because the `ParamValue value` is set from the received param from the autopilot which has the correct type and the `work->param_value` is either `int or float` since this is done while creating the `WorkItem work` in:

https://github.com/mavlink/MAVSDK/blob/9b4b79cbdfca361fab9a528ba78e81376503cbd0/src/mavsdk/core/system_impl.cpp#L770

Do we really need to do this, could not we infer this directly from the incoming data type?
In that case we could have a single api for getting and setting param irrespective for the data type and this would also make it generic and support all data types listed in the mavlink specification.

Looking forward to getting feedback on this.

ykhedar, 2022-02-07:
I just did a dirty hack to implement option B and the changes are in the following commit: 

https://github.com/ykhedar/MAVSDK/commit/12eb9a7358aa2e2d870231735939a7db0af7a9ba

(Also contains a new example to test the param read for ardupilot.)

With this I am able to read all params one by one using `GetInt` and also together using `GetAllParams`. The only catch is, I am currently simply converting the int types to int32 and then using the existing Param api (`GetInt`) to read the params.

ykhedar, 2022-02-09:
Hi, i did some more work to make the param set for all ardupilot params work and the changes required are in the commit:
https://github.com/ykhedar/MAVSDK/commit/b598ab169e30c98c21772bce2beaf9cfdc95c115

Most significant changes are:
1. Addition of SetIntParamRaw() and modifying IntParam to include the param type to the param.proto file. This is done because the knowledge of the param type needs to be sent to the Param plugin to convert the int type to the float type.
2. In function:
`float MAVLinkParameters::ParamValue::get_4_float_bytes() const` 

Adding if condition for int8, int16 types and using direct conversion to float type instead of using the `reinterpret_cast` as used for the int32 type. I think this is the main difference between how the PX4 and APM implementation for parameters differ as specified in:

https://mavlink.io/en/services/parameter.html#ardupilot

The trouble with the `reinterpret_cast` based conversion is that at the end the value 0 is set in the APM autopilot. Not sure if this is due to some other reason but when change the conversion to `std::stof` then i get the correct param set behaviour,

My guess is that the function 
`float MAVLinkParameters::ParamValue::get_4_float_bytes() const` 
needs to be different for PX4 and APM. I am not sure how best to implement that.

Looking forward to getting feedback on this and making the changes more general/robust for an eventual pull request.

ykhedar, 2022-02-09:
Continuing my investigations on MAVSDK support for APM, I made one more commit with mostly examples to test the info and telemetry plugin.

The commit is here: https://github.com/ykhedar/MAVSDK/commit/4bb5e42f51953f6b33ed84a3c066162b01dc18eb
and the results are as follows:

 ## Info  plugin

1. GetFlightInformation   (time_boot_us, flight_uid)

[02:20:12|Warn ] command unsupported (528). (mavlink_commands.cpp:187)
[02:20:13|Warn ] command unsupported (528). (mavlink_commands.cpp:187)
[02:20:14|Warn ] command unsupported (528). (mavlink_commands.cpp:187)

**The command fails for: MAV_CMD_REQUEST_FLIGHT_INFORMATION, probably because we are testing against a sitl and it does not have this info?**

2. GetIdentification            (hardware_uid, legacy_uid)

**Works.**

3. GetProduct                       (vendor_id, name, product_id, name)

Vendor: undefined 
Product: undefined

This is due to the following functions has only one product defined.

```
const std::string InfoImpl::vendor_id_str(uint16_t vendor_id)
const std::string InfoImpl::product_id_str(uint16_t product_id)
```

4. GetVersion                        (Total 11 parameters)

**Woks**

## Telemetry plugin

Only tested a limited telemetry items as listed in the files:
[Test file](https://github.com/ykhedar/MAVSDK/blob/4bb5e42f51953f6b33ed84a3c066162b01dc18eb/examples/apm_info_telemetry/apm_info_telemetry.cpp)

Subscribe* and SetRate* commands work for these except the following:

```
105: MAVLINK_MSG_ID_HIGHRES_IMU, Setting set_rate_imu failed: Command Denied
115: MAVLINK_MSG_ID_HIL_STATE_QUATERNION, Setting set_rate_ground_truth failed: Command Denied
331: MAVLINK_MSG_ID_ODOMETRY, Setting set_rate_odometry failed: Command Denied
340: MAVLINK_MSG_ID_UTM_GLOBAL_POSITION, Setting set_rate_unix_epoch_time failed: Command Denied
Setting set_rate_rc_status failed: Unsupported -> System status is usually fixed at 1 Hz 
```

Probably because these messages are from an external peripheral (sensor, companion computer etc. to Autopilot)
Next up is the Actions plugin.

julianoes, 2022-02-10:
@ykhedar thanks a lot for your investigation. And sorry for not answering earlier, I've been very busy.

> 1. Addition of SetIntParamRaw() and modifying IntParam to include the param type to the param.proto file. This is done because the knowledge of the param type needs to be sent to the Param plugin to convert the int type to the float type.

So I see two ways here:
A) Add the type, as you suggest.
B) Always download all params on startup to then have the type cached when the function is used. This would take longer but make the API easier. What do you think?

> 2. In function: float MAVLinkParameters::ParamValue::get_4_float_bytes() const

That's crazy, I can almost not believe that this implementation is different, that's hilarious, and sad.

> The implementation uses the same message types and workflow. The main difference is that ArduPilot encodes data within parameter message value fields (floats) using a normal C cast rather than a byte-wise copy (and expects incoming messages to use the same encoding). This can result in significant rounding errors.

So I don't actually understand this explanation. If you just cast a float to an int, you will get the rounded (and floored) int number, however, you're suggesting that ArduPilot puts the float value as a 4 byte string and uses `stof`?

I guess whatever is true, we need to handle it correctly on the MAVSDK side, so it matches. And then make sure to have a standard going forward that defines this properly.

ykhedar, 2022-02-20:
> So I see two ways here:
A) Add the type, as you suggest.
B) Always download all params on startup to then have the type cached when the function is used. This would take longer but make the API easier. What do you think?

I think option B is better and I implemented this as a test.

> So I don't actually understand this explanation. If you just cast a float to an int, you will get the rounded (and floored) int number, however, you're suggesting that ArduPilot puts the float value as a 4 byte string and uses stof?

What i understood from the mavlink documentation and the source code at the following line in APM GCS:

[APM GCS](https://github.com/ArduPilot/ardupilot/blob/c83774a7cd83929b90c772beec268d0fc1bf94d2/libraries/GCS_MAVLink/GCS_Param.cpp#L78)

is that before packing the value of the parameter to be set, the value is directly cast into float. Since APM does the type checking based on parameter store on board the autopilot sending the correct type information is also not necessary. So I did the same in my tests. 

More information is coming up in a new pull request pertaining to all of the above changes and also related to missions as well.

julianoes, 2022-03-02:
Sorry for the delay. I will get back to you on this in the next days.

closed 2022-06-30 by ykhedar