mavlink/MAVSDK #1684: undefined reference with debian11_armv7 on RPi

Katawann, 2022-01-20:
Back again :)

So the package installed properly but I built an example and finish with undefined reference issue:
```
$ cmake --build build -j4
Scanning dependencies of target takeoff_and_land
[ 50%] Building CXX object CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o
[100%] Linking CXX executable takeoff_and_land
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `stat64@GLIBC_2.33'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::__cxx11::path::_List::_Impl_deleter::operator()(std::filesystem::__cxx11::path::_List::_Impl*) const@GLIBCXX_3.4.26'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&, std::error_code&)@GLIBCXX_3.4.26'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_detach@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::__cxx11::path::_List::_List()@GLIBCXX_3.4.26'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `__libc_single_threaded@GLIBC_2.32'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_join@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()@GLIBCXX_3.4.26'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__atomic_futex_unsigned_base::_M_futex_wait_until_steady(unsigned int*, unsigned int, bool, std::chrono::duration<long long, std::ratio<1ll, 1ll> >, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> >)@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `stat@GLIBC_2.33'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_create@GLIBC_2.34'
/usr/bin/ld: /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_cond_clockwait@GLIBC_2.34'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/takeoff_and_land.dir/build.make:85: takeoff_and_land] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/takeoff_and_land.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

To reproduce:
```
wget https://github.com/mavlink/MAVSDK/releases/download/v1.0.7/libmavsdk-dev_1.0.7_debian11_armv7.deb
sudo dpkg -i libmavsdk-dev_1.0.7_debian11_armv7.deb
git clone https://github.com/mavlink/MAVSDK.git
cd MAVSDK/examples/
cd takeoff_and_land/
cmake -Bbuild -H.
cmake --build build -j4
```

Information regarding my rpi:

```
$ uname -a
Linux 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

JonasVautherin, 2022-01-20:
Which version of glibc do you have on that rpi?

Katawann, 2022-01-20:
Thank you for the quick answer !

```
$ ldd --version
ldd (Debian GLIBC 2.28-10+rpi1) 2.28
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

By the way, I have the same issue with a docker with Ubuntu18.04/arm/v7

julianoes, 2022-01-20:
Try v1.0.8.

julianoes, 2022-01-20:
And then I expect you to get `libmavsdk.so.1 not found` which means you have to use armv6 instead of armv7.

julianoes, 2022-01-20:
It looks like the 32bit images of Raspberry Pi OS are made backwards compatible for RPi 1, so armv6.

Katawann, 2022-01-20:
Indeed ! 

The version 1.0.8 fixed the build issue of undefined reference, but with `libmavsdk.so.1` not found as you said:
```
$ ./takeoff_and_land 
./takeoff_and_land: error while loading shared libraries: libmavsdk.so.1: cannot open shared object file: No such file or directory
```

but using the armv6 I can run the example. 

Thank you for the great support !!!

closed 2022-01-20 by Katawann

julianoes, 2022-01-20:
Thanks for following up, and glad it's now working!