mavlink/MAVSDK #1685: undefined reference with debian11_armv7 on docker Ubutun18.04 linux/arm/v7

Katawann, 2022-01-20:
The version 1.0.8 of MAVSDK fixed the undefined reference of the debian install for the RPi, but not for a docker container. I am so close to successfully launching my MAVSDK application from a docker container :)

With v1.0.8, I have the following error with the package _debian11_armv6_ and _debian11_armv7_ 
```
#9 46.48 -- Configuring done
#9 46.51 -- Generating done
#9 46.52 -- Build files have been written to: /MAVSDK/examples/takeoff_and_land/build
#9 47.16 Scanning dependencies of target takeoff_and_land
#9 47.33 [ 50%] Building CXX object CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o
#9 55.07 [100%] Linking CXX executable takeoff_and_land
#9 55.64 /usr/lib/libmavsdk.so.1.0.8: undefined reference to `fcntl64@GLIBC_2.28'
#9 55.64 /usr/lib/libmavsdk.so.1.0.8: undefined reference to `fcntl@GLIBC_2.28'
#9 55.65 collect2: error: ld returned 1 exit status
#9 55.67 CMakeFiles/takeoff_and_land.dir/build.make:95: recipe for target 'takeoff_and_land' failed
#9 55.67 make[2]: *** [takeoff_and_land] Error 1
#9 55.67 CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/takeoff_and_land.dir/all' failed
#9 55.67 make[1]: *** [CMakeFiles/takeoff_and_land.dir/all] Error 2
#9 55.67 Makefile:83: recipe for target 'all' failed
#9 55.67 make: *** [all] Error 2
```

With v1.0.7, I had even more error:
```
#10 45.88 -- Configuring done
#10 45.91 -- Generating done
#10 45.92 -- Build files have been written to: /MAVSDK/examples/takeoff_and_land/build
#10 46.60 Scanning dependencies of target takeoff_and_land
#10 46.78 [ 50%] Building CXX object CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o
#10 54.79 [100%] Linking CXX executable takeoff_and_land
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `stat64@GLIBC_2.33'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `fcntl64@GLIBC_2.28'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::__cxx11::path::_List::_Impl_deleter::operator()(std::filesystem::__cxx11::path::_List::_Impl*) const@GLIBCXX_3.4.26'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&, std::error_code&)@GLIBCXX_3.4.26'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_detach@GLIBC_2.34'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::__cxx11::path::_List::_List()@GLIBCXX_3.4.26'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `__libc_single_threaded@GLIBC_2.32'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_join@GLIBC_2.34'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `fcntl@GLIBC_2.28'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()@GLIBCXX_3.4.26'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__atomic_futex_unsigned_base::_M_futex_wait_until_steady(unsigned int*, unsigned int, bool, std::chrono::duration<long long, std::ratio<1ll, 1ll> >, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> >)@GLIBCXX_3.4.29'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `stat@GLIBC_2.33'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_create@GLIBC_2.34'
#10 55.39 /usr/lib/libmavsdk.so.1.0.7: undefined reference to `pthread_cond_clockwait@GLIBC_2.34'
#10 55.40 collect2: error: ld returned 1 exit status
#10 55.41 CMakeFiles/takeoff_and_land.dir/build.make:95: recipe for target 'takeoff_and_land' failed
#10 55.41 make[2]: *** [takeoff_and_land] Error 1
#10 55.41 CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/takeoff_and_land.dir/all' failed
#10 55.41 make[1]: *** [CMakeFiles/takeoff_and_land.dir/all] Error 2
#10 55.42 Makefile:83: recipe for target 'all' failed
#10 55.42 make: *** [all] Error 2
```

Dockerfile to reproduce:
```
FROM ubuntu:18.04

ARG TARGETPLATFORM

# Install dependencies
RUN apt-get update \
    && apt-get -y --quiet --no-install-recommends install \
        build-essential \
        ca-certificates \ 
        git \ 
        sudo \ 
        wget \
        cmake \
    && apt-get -y autoremove \
    && apt-get clean autoclean \
    && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*

# Install MAVSDK prebuilt C++ library
ENV MAVSDK_VERSION=1.0.7
RUN if [ "$TARGETPLATFORM" = "linux/amd64" ]; then MAVSDK_PKG=ubuntu18.04_amd64; \
    elif [ "$TARGETPLATFORM" = "linux/arm/v7" ]; then MAVSDK_PKG=debian11_armv7; \
    else MAVSDK_PKG=ubuntu18.04_amd64; fi \
    && wget https://github.com/mavlink/MAVSDK/releases/download/v${MAVSDK_VERSION}/libmavsdk-dev_${MAVSDK_VERSION}_${MAVSDK_PKG}.deb \
    && sudo dpkg -i libmavsdk-dev_${MAVSDK_VERSION}_${MAVSDK_PKG}.deb \
    && sudo rm libmavsdk-dev_${MAVSDK_VERSION}_${MAVSDK_PKG}.deb

# Run MAVSDK takeoff example
RUN git clone https://github.com/mavlink/MAVSDK.git --recursive \
    && cd MAVSDK/examples/takeoff_and_land/ \
    && cmake -Bbuild -H. \
    && cmake --build build
```

Katawann, 2022-01-20:
glibc is version 2.27, not sure if it fails because of cross-compilation...

```
$ ldd --version
ldd (Ubuntu GLIBC 2.27-3ubuntu1.4) 2.27
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

CY-1992, 2022-01-20:
Change your base your image to 
`FROM  arm32v7/ubuntu:20.04
`

julianoes, 2022-01-21:
> #9 55.64 /usr/lib/libmavsdk.so.1.0.8: undefined reference to `fcntl64@GLIBC_2.28'
> #9 55.64 /usr/lib/libmavsdk.so.1.0.8: undefined reference to `fcntl@GLIBC_2.28'

So [dockcross-lts](https://github.com/dockcross/dockcross/blob/77775208bfea853ba2d34e53f525369293dfaeff/linux-armv6-lts/crosstool-ng.config#L434) is based on glibc 2.28 which seems to [match Debian Buster](https://packages.debian.org/source/buster/glibc), and as you found Ubuntu 18.04 is on glibc 2.27, so just too low.

Any chance you could build on top of Ubuntu 20.04 or Debian Buster (or even Bullseye)?

julianoes, 2022-01-21:
For your cmake error: this works for me on Debian on a RPi... What's your cmake version?

Katawann, 2022-01-21:
I updated the host to Ubuntu 20.04 but I still get the same errors. Below a resume of my tests. Hope this help if there is anything to do to fix it, but at least I have one solution where I can build my docker !

- From host `Ubuntu 18.04 x86_64  glibc:2.27 cmake:3.22.1` cross-compilation with `buildx` to `linux/arm/v7`
  - Target platform 
     - Ubuntu18.04: **_failed_** with `undefined reference to fcntl@GLIBC_2.28'` with `glibc 2.27` and `cmake:3.10.2`
     - Ubuntu20.04: **_failed_** with REMOVE_ITEM [error above](https://github.com/mavlink/MAVSDK/issues/1685#issuecomment-1017812435) 
- From host `Ubuntu 20.04 x86_64 glibx:2.31 cmake:3.22.1` cross-compilation with `buildx` to `linux/arm/v7`
  - Target platform 
     - Ubuntu18.04: **_failed_** with `undefined reference to fcntl@GLIBC_2.28'` with `glibc 2.27` **even if host glibc is now version 2.31**
     - Ubuntu20.04: **_failed_** same error than with Ubuntu 18.04 failed with failed with REMOVE_ITEM [error above](https://github.com/mavlink/MAVSDK/issues/1685#issuecomment-1017812435) 
- From host `Raspberry Pi 3B+ Debian Buster 4.19 armv7l glibx:2.28 cmake:3.16.3` with `docker build` directly
  - Target platform 
     - Ubuntu18.04: **_failed_** with `undefined reference to fcntl@GLIBC_2.28` with `glibc 2.27` and `cmake:3.10.2`
     - Ubuntu20.04: **_failed_** new GPG error `E: The repository 'http://ports.ubuntu.com/ubuntu-ports focal InRelease' is not signed.`
- From host `Raspberry Pi 3B+ Debian Bullseye 5.10 armv7l glibx:2.28 cmake:3.16.3` with `docker build` directly
  - Target platform 
     - Ubuntu18.04: **_failed_** with `undefined reference to fcntl@GLIBC_2.28` with `glibc 2.27` and `cmake:3.10.2`
     - Ubuntu20.04: **succeed !!!**

It would be great to fix this REMOVE_ITEM issue so that I can do cross-compilation but already happy to have one solution working :)

julianoes, 2022-01-21:
It works for me with the `Dockerfile` below. What am I doing wrong?

```
FROM arm32v7/ubuntu:20.04

ENV TARGETPLATFORM=linux/arm/v7

ENV DEBIAN_FRONTEND=noninteractive

# Install dependencies
RUN apt-get update \
    && apt-get -y --quiet --no-install-recommends install \
        build-essential \
        ca-certificates \ 
        git \ 
        sudo \ 
        wget \
        cmake \
    && apt-get -y autoremove \
    && apt-get clean autoclean \
    && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*

# Install MAVSDK prebuilt C++ library
ENV MAVSDK_VERSION=1.0.8
RUN if [ "$TARGETPLATFORM" = "linux/amd64" ]; then MAVSDK_PKG=ubuntu18.04_amd64; \
    elif [ "$TARGETPLATFORM" = "linux/arm/v7" ]; then MAVSDK_PKG=debian11_armv7; \
    else MAVSDK_PKG=ubuntu18.04_amd64; fi \
    && wget https://github.com/mavlink/MAVSDK/releases/download/v${MAVSDK_VERSION}/libmavsdk-dev_${MAVSDK_VERSION}_${MAVSDK_PKG}.deb \
    && sudo dpkg -i libmavsdk-dev_${MAVSDK_VERSION}_${MAVSDK_PKG}.deb \
    && sudo rm libmavsdk-dev_${MAVSDK_VERSION}_${MAVSDK_PKG}.deb

# Run MAVSDK takeoff example
RUN git clone https://github.com/mavlink/MAVSDK.git --recursive \
    && cd MAVSDK/examples/takeoff_and_land/ \
    && cmake -Bbuild -H. \
    && cmake --build build
```

Katawann, 2022-01-21:
I have no idea... I tried with your `Dockerfile` but still the same issue...

you work with `buildx` for the cross-compilation ?

```
$ docker buildx ls
NAME/NODE    DRIVER/ENDPOINT             STATUS  PLATFORMS
multiarch *  docker-container                    
  multiarch0 unix:///var/run/docker.sock running linux/amd64, linux/386, linux/arm64, linux/riscv64, linux/ppc64le, linux/s390x, linux/arm/v7, linux/arm/v6
default      docker                              
  default    default                     running linux/amd64, linux/386, linux/arm64, linux/riscv64, linux/ppc64le, linux/s390x, linux/arm/v7, linux/arm/v6

$ docker buildx build . --platform=linux/arm/v7 -t mavsdk_docker --load
```

I guess something is wrong is my environnement then

julianoes, 2022-01-21:
I didn't know about `buildx`. I use:
```
docker run --rm --privileged multiarch/qemu-user-static --reset -p yes
docker build .
```

Katawann, 2022-01-21:
Ok I see. I still have the issue with your command 😭 

I close the issue, since it is related to who knows what. I will stay with the "no cross platform" solution until I get more time to work on it.

Thanks a lot for your help !

closed 2022-01-21 by Katawann

Katawann, 2022-01-23:
Well, still could not fix that issue but for those who would come across this page: the `REMOVE_ITEM` issue happen on _Ubuntu20.04_, but I succeed to build on _Ubuntu21.04_, _Debian10_ and _Debian11_