mavlink/MAVSDK #1688: Windows Executable - missing mavsdk.dll

Tzhao7, 2022-01-22:
Hi,
After having build issues several days earlier, I managed to successfully build and install MAVSDK. However, after creating and building a cmake C++ project and running the executable, I was given this error:
[image]
I think my CMakeLists.txt is fine but this is what it looks like:
[image]
Want to note that I did a system-wide install and I'm not sure if this should be done on Windows.

JonasVautherin, 2022-01-22:
How did you build and install it? Can you copy the commands here?

Tzhao7, 2022-01-22:
These are the commands I used:
git clone https://github.com/mavlink/MAVSDK.git
git submodule update --init --recursive
cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
runas cmake --build build/default --target install

JonasVautherin, 2022-01-23:
After the build, in your `MAVSDK` directory, can you search for `mavsdk.dll`? Not in `C:\Program Files`, but really in the build folder (so `path/to/MAVSDK/build/default`).

Maybe the system install is not installing the DLL correctly...

Tzhao7, 2022-01-23:
mavsdk.dll is in build/default of the MAVSDK directory. Is the MAVSDK directory supposed to be inside my C++ project? That could be why it's returning this error.

JonasVautherin, 2022-01-23:
> Is the MAVSDK directory supposed to be inside my C++ project? That could be why it's returning this error.

No. The right files from `build/default` should be installed on your system (in `C:\Program Files`) when you run `cmake --build build/default --target install`. It seems to me that we don't install `mavsdk.dll` correctly on Windows, which ends up in your error.

Tzhao7, 2022-01-23:
Should I try a local install and see if the mavsdk.dll can be found there?

JonasVautherin, 2022-01-23:
[This](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/CMakeLists.txt#L94) is (I believe) where we say where the library should be installed. On Linux it gets into `lib/` (more specifically `lib/libmavsdk.so`, `lib/libmavsdk.so.0`, `lib/libmavsdk.so.0.50.1`).

Somehow it seems like on Windows the `.dll` (which should be the equivalent to the `.so` on Linux) is not installed correctly.

Tzhao7, 2022-01-23:
So after doing a local install I was still not able to find mavsdk.dll within the ### install directory. Only see it in ### build/default.

JonasVautherin, 2022-01-23:
Could you try to debug it? For instance, you could try to print `${CMAKE_INSTALL_LIBDIR}` [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/CMakeLists.txt#L94) to see if it looks correct on Windows. With something like:

```
message(FATAL_ERROR "CMAKE_INSTALL_LIBDIR: ${CMAKE_INSTALL_LIBDIR}")
```

Tzhao7, 2022-01-23:
I modified the CMakeLists.txt to include that line of code and was given this: ```CMake Error at src/mavsd/core/CMakeLists.txt:98 (message): CMAKE_INSTALL_LIBDIR: lib```

JonasVautherin, 2022-01-23:
That looks fine to me :thinking:

Tzhao7, 2022-01-24:
Do you think the previous releases have this problem? It seems like other Windows users haven't had this problem or have had the issue resolved.

julianoes, 2022-01-27:
https://cmake.org/cmake/help/v3.0/variable/CMAKE_INSTALL_PREFIX.html

What about passing the directory where you need/want it?

```
cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -H.
cmake --build build/default --target install
```

JonasVautherin, 2022-01-27:
But it seems like it's installing "stuff" already, just not that one dll :thinking:. Do you think that a local install would help?

julianoes, 2022-01-27:
Is it's installing it into `C:\Program Files\lib`? That would be a bit awkward.

JonasVautherin, 2022-01-27:
I was assuming something like `C:\Program Files\mavsdk\lib`, but I don't know what CMake does on Windows really :sweat_smile:. However from this:

> So after doing a local install I was still not able to find mavsdk.dll within the install directory. Only see it build/default.

my understanding was that `mavsdk.dll` is not installed correctly, even locally

julianoes, 2022-01-27:
> mavsdk.dll within the install directory

The question is which one that is.

Tzhao7, 2022-01-31:
Although this is technically incorrect, I was able to temporarily fix the dependency problem by moving a copy of mavsdk.dll into the same directory as my project executable.

julianoes, 2022-02-01:
@Tzhao7 ok, do you have any suggestions what we should change? I really don't know how linking works properly on Windows.

JonasVautherin, 2022-02-01:
> Also I noticed that the installed library found in C:\Program Files only contains mavsdk.lib and does not contain any .dll.

> mavsdk.dll is in build/default of the MAVSDK directory

Given [this information](https://stackoverflow.com/questions/913691/dll-and-lib-files-what-and-why):

> There are static libraries (LIB) and dynamic libraries (DLL) - but note that .LIB files can be either static libraries (containing object files) or import libraries (containing symbols to allow the linker to link to a DLL).

and assuming that the `mavsdk.lib` in this case is an import library, then my hunch is that `mavsdk.dll` should be installed next to `mavsdk.lib`, but it is not. Currently we install it [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/CMakeLists.txt#L92-L96).

JonasVautherin, 2022-02-01:
Where is `mavsdk.lib`? `C:\Program Files\mavsdk\lib\mavsdk.lib`?

Tzhao7, 2022-02-01:
Yeah it is in `C:\Program Files (x86)\mavsdk_superbuild\lib\mavsdk.lib`. I copied the mavsdk.dll from the cloned MAVSDK to the lib folder and the error continued to occur.

JonasVautherin, 2022-02-01:
@Tzhao7: Would you mind trying to build and install again from the [fix-windows-install](https://github.com/mavlink/MAVSDK/tree/fix-windows-install) branch and tell me if that changes anything? I would expect `mavsdk.dll` to be installed in `:\Program Files (x86)\mavsdk_superbuild\bin\mavsdk.dll`.

Tzhao7, 2022-02-02:
After building and installing from the fix-windows-install branch I was able to find `mavsdk.dll` in `:\Program Files (x86)\mavsdk_superbuild\bin\mavsdk.dll`. However after removing `mavsdk.dll` from my Visual Studio project I started to received the dependency error again.

JonasVautherin, 2022-02-02:
Grmbl, I don't know where the dll should go, and if we are missing something else. My understanding is that the import library (`mavsdk.lib`) should have some kind of reference to `mavsdk.dll`. 

Do you know how to inspect `mavsdk.lib`? I don't know, something like `dumpbin.exe -headers mavsdk.lib` maybe? Just to see if there is a path or something there...

At least now it installs the DLL, so I think that's progress.

Tzhao7, 2022-02-02:
I ran `dumpbin.exe -headers mavsdk.lib` and directed the output to a text file. I wasn't able to find an explicit path but there were references to `mavsdk.dll`. Here's a snippet of what one output paragraph looks like (rest of the output had similar-looking paragraphs):
`Version      : 0
  Machine      : 8664 (x64)
  TimeDateStamp: E1485C25
  SizeOfData   : 00000094
  DLL name     : mavsdk.dll
  Symbol name  : ?get_option@CameraImpl@mavsdk@@AEAA?AW4Result@Camera@2@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEAUOption@42@@Z (private: enum mavsdk::Camera::Result __cdecl mavsdk::CameraImpl::get_option(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,struct mavsdk::Camera::Option &))
  Type         : code
  Name type    : name
  Hint         : 12295
  Name         : ?get_option@CameraImpl@mavsdk@@AEAA?AW4Result@Camera@2@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEAUOption@42@@Z`

JonasVautherin, 2022-02-02:
Can you try to move `mavsdk.dll` from `bin/` into `lib/`, so that it's next to `mavsdk.lib`? That was not working before, and I don't think that my PR changed anything about that, but just in case.

What I don't get is: assuming that it finds `lib/mavsdk.lib`, and then tries to find `mavsdk.dll` because of that `DLL name` field, then why doesn't it find it next to `mavsdk.lib`?

My other question would be: does it actually use `lib/mavsdk.lib`? What if you remove it entirely, does it still complain about "The code execution cannot proceed because mavsdk.dll was not found"?

Tzhao7, 2022-02-03:
After moving `mavsdk.dll` from `bin/` into `lib/`, I was given the following error message when compiling the CMakeLists.txt:
[image]
It seems like `lib/mavsdk.lib` is looking for `bin/mavsdk.dll`. Removing `lib/mavsdk.lib` entirely returned the following error message: 
[image]
However, as you said previously, when I tried moving `mavsdk.dll` next to `mavsdk.lib` before, I received those dependency errors. I'm not exactly sure what is going on either haha.

JonasVautherin, 2022-02-03:
Wait but then it's looking for it now :thinking:. Are you sure you made a clean build with my PR?

julianoes, 2022-02-11:
@Tzhao7 could you please report back if the PR helped (after a clean build)?

Tzhao7, 2022-02-11:
Sorry for the delay...I made a clean build with the PR and moved mavsdk.dll into `lib/` but the runtime error still appears.

julianoes, 2022-02-19:
@Tzhao7 ok do you have a suggestion what we could do to fix this? @JonasVautherin and I are both not using Windows normally, so we don't know how cmake on Windows is usually done.

Tzhao7, 2022-02-19:
To be honest I am also not a native Windows user and do not know how cmake works either. The only way I was able to resolve the dependency error was by placing mavsdk.dll next to my project executable and I don't think this is a viable solution. I wonder if other Windows users had this problem after building and installing the SDK.

julianoes, 2022-03-02:
> by placing mavsdk.dll next to my project executable and I don't think this is a viable solution.

This is usually what I do. :smirk:

PR #1691 (2022-03-15): Set RUNTIME DESTINATION to fix Windows install
Trying a fix for https://github.com/mavlink/MAVSDK/issues/1688

Resolves #1688
closed 2022-03-15 by julianoes