mavlink/MAVSDK #1689: Error in mavsdk/plugin_base.h

slaghuis, 2022-01-24:
I have code that compiled and ran under version 0.30 of MAVSDK.  I did experience some connection inconsistencies with my PIXHAWK MINI and decided to upgrade to the latest release of MAVSDK, as Version 0.33 brought about improvements in this area.  I removed the MAVSDK::mavsdk_action, etc libraries from my CMakeLists.txt and now I get the following error when compiling my code.
`In file included from /usr/local/include/mavsdk/plugins/info/info.h:16,
                 from /home/eric/ros_ws/src/drone/src/drone_node.cpp:46:
/usr/local/include/mavsdk/plugin_base.h:25:61: error: expected constructor, destructor, or type conversion before ‘;’ token
   25 | template<class... Ts> overloaded(Ts...) -> overloaded<Ts...>;
      |                                                             ^
make[2]: *** [CMakeFiles/base_node_lib.dir/build.make:63: CMakeFiles/base_node_lib.dir/src/drone_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:136: CMakeFiles/base_node_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
`
I did not change anything else in my code.  How do I debug this? I get the error if I include any mavsdk header files.  

I am running Ubuntu 20.04 64 bit (Both on a Virtual Box VM and on a Raspberry Pi 4)

Any advice would be appreciated.

JonasVautherin, 2022-01-24:
Can you try building an example with the latest MAVSDK release, to confirm that this works?

slaghuis, 2022-01-24:
I have upgraded my MAVSDK to version 1.0.8-2-g5806dc04, that I manually built by downloading the latest code from GitHub.  MAVSDK compiles and installs successfully. My result remains the same, I am afraid.  

I was able to compile the MAVSDK/examples/battery example.  

Maybe the error is in my code.  Let me first have another look at my code.

slaghuis, 2022-01-24:
I wrote a simple ROS2 node that does nothing really.  It connects to the flight controller.  This code compiles and functions.  The moment I want to add actions or telemetry to this, I need to include the relevant header file.  Then  get the same error message as above. 

Is there perhaps a minimum requirement on the C compiler that I need to use?  C++ 20 perhaps?  My CMakeLists.txt states:

`project(drone_mavsdk)

if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
find_package(MAVSDK REQUIRED)

add_executable(drone_node src/drone_node.cpp)
ament_target_dependencies(drone_node rclcpp std_msgs)

target_link_libraries(drone_node
  MAVSDK::mavsdk
)

install(TARGETS
  drone_node
  DESTINATION lib/${PROJECT_NAME})

ament_package()`

slaghuis, 2022-01-24:
I changed the CMakeLists.txt to state
`
if(NOT CMAKE_CXX_STANDARD)
set(CMAKE_CXX_STANDARD 17)
endif()
`
And it works!

Sorry to have bothered the forum.

closed 2022-01-24 by slaghuis

JonasVautherin, 2022-01-24:
Good catch!

shivamydv2001, 2023-10-17:
I got  this error while  I am compiling the code of battery, please guide me to solve this , Thank you  

 In file included from battery.cpp:7:
/usr/include/mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h:9:10: fatal error: plugin_base.h: No such file or directory
    9 | #include "plugin_base.h"

julianoes, 2023-10-17:
@shivamydv2001 this should be resolved in the meantime. Which version are you trying to build?

julianoes, 2023-10-17:
@shivamydv2001 let's follow up in https://github.com/mavlink/MAVSDK/issues/2156 instead.

shivamydv2001, 2023-10-18:
@julianoes can you help me to solve this issue and version is version is v1.32 gimbal-STorm 32 BGC and ubuntu20.04,  
In file included from battery.cpp:7:
/usr/include/mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h:9:10: fatal error: plugin_base.h: No such file or directory
9 | #include "plugin_base.h"