mavlink/MAVSDK #1692: MAVSDK-Python / examples / camera.py doesn't work

takata1021, 2022-02-04:
I'm running typhoon_h480 or plane_cam from the Gazebo simulator and running the MavSDK camera.py.
The camera class will result in an error.
You can shoot with the camera from QGroundControl.
Is it possible to operate the camera from the Mav SDK while the Gazebo simulator is running? Please tell me how to do it.
Is this a bug?

JonasVautherin, 2022-02-04:
I don't think that Gazebo SITL handles the camera messages, does it?

julianoes, 2022-02-04:
It would be nice if this was supported by PX4 SITL Gazebo. I don't think it is in the current state.

takata1021, 2022-02-06:
thank you.
It means that the PX4 SITL gazebo does not support camera operation using the Mav SDK.
It was helpful.

takata1021, 2022-02-06:
QGroundControl supports photo ops and video streaming with access to gazebo cameras on UDP port 5600.
Isn't this communicating with a given UDP port using MAVLinkAPI?
We believe the MavSDK can do the same.

takata1021, 2022-02-08:
thank you.

In other words, is it okay that the camera class of the Mav SDK works correctly when connected to the actual drone, but the camera class cannot be used when connected to the Gazebo simulator (SITL)?
Is it okay that the camera class of MavSDK does not support SITL?

Jaeyoung-Lim, 2022-02-08:
@julianoes @JonasVautherin I think I am missing a bit of context - what is missing on the SITL side? If it is useful, I can also add anything that is missing

takata1021, 2022-02-16:
What's happening in this case?
I would like an answer.

JonasVautherin, 2022-02-17:
How do you know it's supported by SITL? I don't think SITL sends those messages. MAVSDK supports them (if you use a real drone, you'll see that you can control the camera), but if SITL does not, there is nothing MAVSDK can do, right?

takata1021, 2022-02-21:
SITL doesn't support cameras, right?
got it.
If so, please mention that in the user guide.

julianoes, 2022-03-02:
It does support cameras with a live stream but not camera settings, I guess.

If you feel like the user guide is misleading, then just open a pull request to it (https://github.com/PX4/PX4-user_guide/) to improve it as you think would be better.

takata1021, 2022-03-02:
Does SITL support typhoon_h480 (video story camera) but not Mav SDK?
 I confirmed that I can shoot from QGrroundControl using typhoon_h480 with SITL. However, it could not be operated from the Mav SDK.

julianoes, 2022-03-03:
When you take a picture from QGroundControl it just grabs a picture from the stream and saves it, as far as I remember.

OliverHeilmann, 2023-09-24:
To chip in here, I did manage to connect to a camera from the PX4 SITL environment:

First run the simulator:
```console
oliver@ubuntu:~/PX4-Autopilot$ make px4_sitl gazebo_typhoon_h480
```

Then I ran the camera example code (but added a few extra prints):
```console
oliver@ubuntu:~/MAVSDK/examples/camera$ ./build/camera "udp://:14550"
[05:54:50|Info ] MAVSDK version: v1.4.0-532-g48f1ad03 (mavsdk_impl.cpp:23)
Waiting to discover system...
[05:54:50|Info ] New system on: 127.0.0.1:18570 (with system ID: 1) (udp_connection.cpp:192)
[05:54:50|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:701)
[05:54:50|Debug] Component Autopilot (1) added. (system_impl.cpp:379)
[05:54:50|Debug] Component Gimbal (154) added. (system_impl.cpp:379)
[05:54:50|Info ] New system on: 127.0.0.1:14530 (with system ID: 1) (udp_connection.cpp:192)
[05:54:50|Debug] Component Camera 1 (100) added. (system_impl.cpp:379)
[05:54:50|Debug] Discovered 3 component(s) (system_impl.cpp:548)
Discovered camera
[05:54:50|Warn ] Received ack for not-existing command: 525! Ignoring... (mavlink_command_sender.cpp:279)
[05:54:50|Warn ] Received ack for not-existing command: 521! Ignoring... (mavlink_command_sender.cpp:279)
[05:54:50|Warn ] Received ack for not-existing command: 2000! Ignoring... (mavlink_command_sender.cpp:279)
Received capture info:
  Position: 47.3978, 8.54561, 0.488 m
  Attitude: 0, 0, 0, 0
  Attitude: 0, 0, 0
  Time UTC: 133916000
  Success: 1
  Index: 2
  File URL: 
[05:54:51|Warn ] sending again after 0.500743 s, retries to do: 3  (520). (mavlink_command_sender.cpp:308)
[05:54:51|Warn ] Vehicle type changed (new type: 13, old type: 0) (system_impl.cpp:222)
```

Couple issues though.
1) A camera is only *sometimes* discovered. I had to run the code a few times before I actually connected to a camera. Not sure why this would be?
2) I have no idea where the image is saved to? Another thing, how do I actually access the picture/ video frames from MAVSDK? Is this supported?

Thanks in advance!

JonasVautherin, 2023-09-24:
> A camera is only sometimes discovered.

Hmm that's weird... I think it relies on the heartbeat, so would it mean that the camera heartbeat is unreliable? Just guessing here.

> I have no idea where the image is saved to?

It is not, it delegates that to you. What MAVSDK will give you is the URLs to access the pictures/video streams (see below).

> Another thing, how do I actually access the picture/ video frames from MAVSDK? 

When you take a picture, you receive a [`CaptureInfo` that contains a `file_url`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/camera/camera.proto#L314). If your drone exposes e.g. an HTTP server, that's the URL where you can get the picture. So you want to [`SubscribeCaptureInfo`](https://github.com/mavlink/MAVSDK-Proto/blob/8966cf59029ac17d81654071ca6d3d27f0d8897d/protos/camera/camera.proto#L75) here, or [`ListPhotos`](https://github.com/mavlink/MAVSDK-Proto/blob/8966cf59029ac17d81654071ca6d3d27f0d8897d/protos/camera/camera.proto#L59) to list all of them.

For the video stream, with the [`VideoStreamSettings` you get a `video_stream_uri`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/camera/camera.proto#L324). Typically that would be an `rtsp://` address that you can use to access the video. So you want to use [`SubscribeVideoStreamInfo`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/camera/camera.proto#L71) to get that.

julianoes, 2023-09-26:
I'm working on getting some of th camera/stream functionality developed. Stay tuned. Although for "server" plugins there is not a lot of support in languages other than X..

OliverHeilmann, 2023-10-08:
So this isn't exactly the same as your suggestion, but I managed to access the videostream with GStreamer and OpenCV with the following command when initialising a camera:

```cpp
   // Define the Gstreamer pipeline
    std::string pipeline = "udpsrc port=5600 ! application/x-rtp,encoding-name=H264,payload=96 ! "
                           "rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! appsink";

    // Open the video stream
    camera = cv::VideoCapture(pipeline, cv::CAP_GSTREAMER);
```

The [PX4 docs](https://dev.px4.io/v1.11_noredirect/en/simulation/gazebo.html#video) on video streaming helped a little too.

I now understand that connecting to the camera through the AutoPilot software is not used for streaming but rather for controlling basic camera functionality e.g. take photo etc. (though perhaps also streaming?) Video streaming was a little new to me but the above command should sort PX4 sitl issues out. 

@JonasVautherin, the inconsistent connection is definitely still a bit of an issue in general- perhaps due to heartbeat like you've suggested... not sure though.