mavlink/MAVSDK #1698: Tips on to make a drone follow a virtual line in the air?

reedev, 2022-02-14:
HI,

I would like to make my drone follow a virtual line in the air. I don't have the need to automatically take pictures or anything. Once the drone is on the virtual line (lat, long, alt) I want to go back and forth along this line and still being able to change the heading.

Can someone give me tips and tricks how something like that can be achieved?

Thanks,
Raymond

julianoes, 2022-02-19:
Interesting, thanks for sharing. This would indeed be nice but we don't really have the feature in PX4 yet (I don't know about ArduCopter), and I'm not so sure that we even have mavlink messages to enable it.

Could we make it work with mission items? Or would we want a separate "spline" API? What would that look like?

reedev, 2022-02-20:
Indeed interesting, and will produce smooth motions. To start with I would be ok with following the current mission segments and only go forward. Any thought on that with the current PX4 and MAVSDK? I don’t have a clue at the moment where to start looking so any suggestion is appreciated.

julianoes, 2022-03-02:
So if we are to use the existing mission item API, we would presumably use this:
https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_SPLINE_WAYPOINT
I'm not entirely sure how this spline would work though. It doesn't really seem fully specced.

If we were to add a new message it would be something closer to:
https://mavlink.io/en/messages/common.html#TRAJECTORY_REPRESENTATION_BEZIER
But it's sorta limited to 5 points, so that's not really it either, right?

reedev, 2022-03-03:
Hmm, that is an interesting finding. When I have more time, I will browse the code too in this direction...

I started this thread not because I want to have non-linear lines though. What I would first like to achieve is to follow the line of a mission (current piece wise linear segments are perfectly fine at the moment) and fly along the line forward and backward manually.

I will try to explain it in more detail... At the moment, you set the speed parameter and the mission is using that to fly autonomously forward. Now I would like to have control to make the drone forward or backward along the mission line via a parameter or so. So that would mean that e.g. when the parameter = 0 the drone is keeping it's location, > 0 for going forward along the mission line and < 0 for going backward along the mission line.

And when that's possible one step further (not needed per se but would be nice), is that the heading can be changed with another parameter, when the drone is still following the mission line.