mavlink/MAVSDK #1702: Add set flight mode to public API?

reedev, 2022-02-21:
Hi @julianoes 

I found https://discuss.px4.io/t/is-it-possible-to-set-other-flight-modes-stabilized-hold-etc/17338/2 and would like to follow up here, in the context of MAVSDK.

I was looking for the same functionality and I think I have a use case: when I am in mission mode and I pause the mission, the flight mode is implicitly put into Hold. And that means that the drone will return to the location where it was when pause was issued, and it will turn around too.

I want to go into Position flight mode and accept that the drone is not exactly in the location where the pause cmd was issued. For me, it's not an issue to back up a bit manually after the mission has been paused.

Could you add SystemImpl::set_flight_mode() to System? Or am I missing an important point here?

Thanks, cheers,
Raymond

reedev, 2022-02-21:
Indeed, that is the question. The PX4 spec says that the flight mode should be Hold when a mission is paused: https://docs.px4.io/master/en/flight_modes/mission.html so I guess when the current behaviour is changed it might break for others.

So I guess the suggested behaviour is a MAVSDK specific change. What do you think?

julianoes, 2022-02-22:
This will be fixed in PX4, see: https://github.com/PX4/PX4-Autopilot/pull/19244

reedev, 2022-03-01:
That sounds like an elegant solution. I have been trying this out with my app but encounter build issues with the current PX4. Will contact you via slack...

reedev, 2022-03-01:
Nice, Julian, I tried out PX4 with sitl and the behaviour is now nice, just like you mentioned. Thanks.

closed 2022-03-01 by reedev

keiler2018, 2022-08-31:
APM cannot take off unless the ground station is set to guided  mode. How to solve this???

julianoes, 2022-08-31:
@keiler2018 ArduPilot support is limited, see https://github.com/mavlink/MAVSDK/issues/1568.

We should implement the mode change the same way it is implemented in the offboard plugin for PX4, just with the ArduPilot mode change required, see: 
https://github.com/mavlink/MAVSDK/blob/261e15d2a8687838534eee0cf1fd16f6fb511df6/src/mavsdk/plugins/offboard/offboard_impl.cpp#L56

Even better would be if everything switched to common flight modes: https://github.com/mavlink/rfcs/pull/17