mavlink/MAVSDK #1704: Build on Linux gives error: jsoncpp does not appear to contain CMakeLists.txt

dsmits32, 2022-02-21:
Building on Linux gives an error
```
CMake Error: The source directory "/home/MAVSDK/build/default/third_party/jsoncpp" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
CMake Error at third_party/cmake/build_target.cmake:31 (message):
  /home/MAVSDK/third_party/jsoncpp failed to configure!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:10 (build_target)
```

Following command was used for the build:
`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_MAVSDK_SERVER=ON -DBUILD_SHARED_LIBS=OFF -Bbuild/default -H.`

JonasVautherin, 2022-02-21:
Is it a clean build?

dsmits32, 2022-02-22:
Thank you for your quick reply. Much appreciated.

It's in a Docker image build. So yes its a clean build. (Docker build with --no-cache option).
Here is the Dockerfile, maybe it provides more info:

```
FROM arm64v8/ubuntu:bionic

WORKDIR /home

RUN apt-get update
RUN apt-get install build-essential colordiff git doxygen wget libssl-dev -y
RUN wget https://github.com/Kitware/CMake/releases/download/v3.23.0-rc1/cmake-3.23.0-rc1-linux-aarch64.sh \
    -q -O /tmp/cmake-install.sh \
    && chmod u+x /tmp/cmake-install.sh \
    && mkdir /usr/bin/cmake \
    && /tmp/cmake-install.sh --skip-license --prefix=/usr/bin/cmake \
    && rm /tmp/cmake-install.sh

ENV PATH="/usr/bin/cmake/bin:${PATH}"

RUN git clone https://github.com/mavlink/MAVSDK.git
WORKDIR /home/MAVSDK
RUN git checkout main
RUN git submodule update --init --recursive

RUN cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_MAVSDK_SERVER=ON -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -H.
RUN cmake --build build/default --target install
```

JonasVautherin, 2022-02-22:
Seems like something is wrong with your image. Have you seen the warnings?

```
[Warning] The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64) and no specific platform was requested
```

dsmits32, 2022-02-23:
Thanks again for your help. 
Yes indeed. You have to run an arm64 container on x86 via an emulator:
https://www.stereolabs.com/docs/docker/building-arm-container-on-x86/

But this is only for your info. Since it seems I’m the only one with this problem, it probably has something to do with my container or emulator. It will investigate it further.

JonasVautherin, 2022-02-23:
Did you try running the same without the emulator?

Our CI builds a ton of scenarios, I would be surprised if that was a bug on our side: https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml

dsmits32, 2022-02-24:
The problem only occurs on Ubuntu 18. It does work on Ubuntu 20. We wil continue with 20. 
Thanks for your help.

closed 2022-02-24 by JonasVautherin