mavlink/MAVSDK #1706: undefined MissionRaw XXX in v1.0.8

dzywater, 2022-02-28:
unresolved external symbol "public: __cdecl Json::Value::Value(enum Json::ValueType)" (??0Value@Json@@QEAA@W4ValueType@1@@Z) referenced in function "public: static struct std::pair<enum mavsdk::MissionRaw::Result,struct mavsdk::MissionRaw::MissionImportData> __cdecl mavsdk::MissionImport::parse_json(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?parse_json@MissionImport@mavsdk@@SA?AU?$pair@W4Result@MissionRaw@mavsdk@@UMissionImportData@23@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@4@@Z)

unresolved external symbol "public: __cdecl Json::Value::Value(char const *)" (??0Value@Json@@QEAA@PEBD@Z) referenced in function "private: static class std::optional<class std::vector<struct mavsdk::MissionRaw::MissionItem,class std::allocator<struct mavsdk::MissionRaw::MissionItem> > > __cdecl mavsdk::MissionImport::import_complex_mission_items(class Json::Value const &)" (?import_complex_mission_items@MissionImport@mavsdk@@CA?AV?$optional@V?$vector@UMissionItem@MissionRaw@mavsdk@@V?$allocator@UMissionItem@MissionRaw@mavsdk@@@std@@@std@@@std@@AEBVValue@Json@@@Z)

....omit more.

julianoes, 2022-03-02:
Can you share commands how to reproduce this please? Then I can have a look.

dzywater, 2022-03-03:
> Can you share commands how to reproduce this please? Then I can have a look.

My code is long. But you can reproduce it by following simple lines:

#include <mavsdk/mavsdk.h>
#include <mavsdk/system.h>
#include <mavsdk/plugins/camera/camera.h>
int main(int argc, char** argv)
{
	std::shared_ptr<mavsdk::System> sys;
	auto cam = std::make_shared<mavsdk::Camera>(sys);
	return 0;
}

julianoes, 2022-03-03:
This doesn't look correct:
```
std::shared_ptr<mavsdk::System> sys;
```

Can you try like this instead?
```
#include <memory>
#include <chrono>
#include <thread>
#include <mavsdk/mavsdk.h>
#include <mavsdk/system.h>
#include <mavsdk/plugins/camera/camera.h>

int main(int argc, char** argv)
{
    mavsdk::Mavsdk mavsdk;
    mavsdk.add_udp_connection();    

    // Wait for system to connect via heartbeat.
    std::this_thread::sleep_for(std::chrono::seconds(2));
    
    if (mavsdk.systems().empty()) {
        std::cout << "No system found\n";
        return -1;
    }

    auto system = mavsdk.systems().at(0);

    if (!system->has_camera()) {
        std::cout << "No camera found\n";
        return -1;
    }

    auto camera = mavsdk::Camera{system};
}
```

dzywater, 2022-03-04:
I tested your codes.Same error.
I knew “ std::shared_ptr<mavsdk::System> sys;”  is incorrect. It is just for compilation test not runtime test.

julianoes, 2022-03-06:
Ok, how are you building MAVSDK? When you get mavsdk.lib, do you also see curl and jsoncpp libs? You might be missing them.

dzywater, 2022-03-18:
> Ok, how are you building MAVSDK? When you get mavsdk.lib, do you also see curl and jsoncpp libs? You might be missing them.

I download precompiled zip file directly.

julianoes, 2022-03-23:
Ok, I will have to try that on Windows then. Maybe the libraries in the zip files are missing some dependencies.

julianoes, 2024-05-27:
I'm closing this, I doubt it's still relevant.

closed 2024-05-27 by julianoes