mavlink/MAVSDK #1710: mavsdk::System::has_camera()  return false for gazebo_typhoon_h480

dzywater, 2022-03-03:
For gazebo_typhoon_h480
mavsdk::System::has_gimbal() return true;
mavsdk::System::has_camera() return false;
Actually, gazebo_typhoon_h480 has gimbal and camera.
The QGroundControl can watch camera video.

julianoes, 2022-03-03:
The Typhoon H480 SITL model does not support the camera definition part. We would have to implement that on the PX4 SITL side.

dzywater, 2022-03-04:
Could you know which drone model has a camera. I want to test mavsdk::Camera.
Additionally, I want to ask how QGroundControl get video stream from H480.

julianoes, 2022-03-06:
No SITL model has a camera implementation from what I know. It would have to be implemented, either using camera-manager or the camera server functionality which is currently being added to mavsdk. Sorry about that.