mavlink/MAVSDK #1711: mavsdk::Gimbal::control() exception but mavsdk::Gimbal::subscribe_control() OK

dzywater, 2022-03-03:
Call mavsdk::Gimbal::control() exception.
Call mavsdk::Gimbal::subscribe_control() OK.

some codes:

		gimRet = gimbal->take_control(mavsdk::Gimbal::ControlMode::Primary);
		if (gimRet != mavsdk::Gimbal::Result::Success) { spdlog::error("gimbal.take_control failed with {}", gimRet); return; }
		mavsdk::Gimbal::ControlStatus ctl;//exception when call gimbal->control()
		gimbal->subscribe_control([&ctl](mavsdk::Gimbal::ControlStatus v)->void { ctl = v; });

		//3.ManualGimbal
		for (int k = 0; k < 5; ++k)
		{
			gimRet = gimbal->set_pitch_and_yaw(0, k * 90);
			if (gimRet != mavsdk::Gimbal::Result::Success) spdlog::error("gimbal.set_pitch_and_yaw failed with {}", gimRet);
			string str, prefix = "\t";
			str += fmt::format("{}control_mode: {}\n", prefix, ctl.control_mode);
			str += fmt::format("{}sysid_primary_control: {}\n", prefix, ctl.sysid_primary_control);
			str += fmt::format("{}sysid_secondary_control: {}\n", prefix, ctl.sysid_secondary_control);
			str += fmt::format("{}compid_primary_control: {}\n", prefix, ctl.compid_primary_control);
			str += fmt::format("{}compid_secondary_control: {}\n", prefix, ctl.compid_secondary_control);
			spdlog::info("Gimbal control status\n{}", str);
			this_thread::sleep_for(5s);
		}

		//4.ReleaseControl
		gimRet = gimbal->release_control();
		if (gimRet != mavsdk::Gimbal::Result::Success) spdlog::error("gimbal.release_control failed with {}", gimRet);

JonasVautherin, 2022-03-03:
Which version of MAVSDK are you using? On which platform? Are you able to write a minimal, runnable example that reproduces the bug so that someone can try to fix it?

dzywater, 2022-03-03:
V1.0.8，ubuntu20.04.
On Windows10, System::has_gimbal() returns false.

julianoes, 2022-03-03:
Can you provide the full compile or run output please? I have never seen this before.

dzywater, 2022-03-04:
[image]

julianoes, 2022-03-06:
@dzywater by the way, if you put ``` around your code it gets nicely formatted.

Thanks for sharing the example. I'll try to reproduce this.

julianoes, 2022-03-06:
I can't reproduce this with MAVSDK from the main branch or release 1.0.8.

If you could get a backtrace where it happens, that would be very helpful.
To get a backtrace, the easiest is to run it in gdb:

```gdb --args your_exe```

Then you type `run` and enter, and once it crashes you do `backtrace` and copy that here.

julianoes, 2023-03-13:
Closing as there was no follow up.

closed 2023-03-13 by julianoes