mavlink/MAVSDK #1714: mavsdk::System::has_camera() return false on Windows10

dzywater, 2022-03-04:
On ubuntu20.04, it is workable (mavsdk::System::has_camera() return false ).
On Windows10, it is not workable (mavsdk::System::has_camera() return false ).
[image]
[image]

JonasVautherin, 2022-03-04:
How is that different from https://github.com/mavlink/MAVSDK/issues/1710?

dzywater, 2022-03-04:
> How is that different from #1710?
Use gazebo_typhoon_h480
On ubuntu20.04, mavsdk::System::has_gimbal()  return true, mavsdk::System::has_camera() return false;

On Windows10, mavsdk::System::has_gimbal()  return false, mavsdk::System::has_camera() return false;

That is,

On Ubuntu20.04, gimbal is workable, but camera is not.
On Windows10, both gimbal and camera are not workable.

julianoes, 2022-03-06:
On Windows, which simulation and which model are you using?

dzywater, 2022-03-18:
1.Run gazebo_typhoon_h480 simulator On ubuntu20.04.
2.Map Ubuntu port14540 to Windows port 14540 by MavRouter.
3.On Windows, use MavSDK1.0.8 to program.

julianoes, 2022-03-23:
Aha, my guess is that something with the forwarding goes wrong. What mavlink-router version are you using?

julianoes, 2024-05-27:
No more answer, closing.

closed 2024-05-27 by julianoes