mavlink/MAVSDK #1725: Change libmavsdk third party libs installation path

BestITUserEUW, 2022-03-21:
Hey,
i'm currently trying to build the mavsdk library from source, and include it in my own project but i am having issues with changing the CMAKE_INSTALL_PREFIX for third party libs in mavsdk. I'm building mavsdk with https://github.com/dockcross/dockcross toolchain and the following cmakeList:
cmake_minimum_required(VERSION 3.1)

project(external-MAVSDK)
include(ExternalProject)

list(APPEND CMAKE_ARGS
    "-DCMAKE_INSTALL_PREFIX:PATH=${CMAKE_INSTALL_PREFIX}"
    "-DCMAKE_TOOLCHAIN_FILE:PATH=${CMAKE_TOOLCHAIN_FILE}"
    "-DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}"
    "-DBUILD_SHARED_LIBS=OFF"
    "-DBUILD_TESTS=OFF"
)

message(STATUS "Preparing external project \"MAVSDK\" with args:")
foreach(CMAKE_ARG ${CMAKE_ARGS})
    message(STATUS "-- ${CMAKE_ARG}")
endforeach()

ExternalProject_add(
    MAVSDK
    GIT_REPOSITORY https://github.com/mavlink/MAVSDK.git
    PREFIX MAVSDK
    GIT_TAG v1.0.8
    CMAKE_ARGS "${CMAKE_ARGS}"
)
but the problem is that the third party libs from MAVSDK are being installed in:
`/work/build/third_party/MAVSDK/MAVSDK/src/MAVSDK-build/third_party/install`
instead of where the other libraries being installed in:
`/work/build/third_party/install/include`
I don't know if this is a problem but i am getting the following error when trying to find the MAVSDK package:

  Make Error at /usr/share/cmake-3.21/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  build/third_party/install/lib/cmake/MAVSDK/MAVSDKConfig.cmake:7 (find_dependency)
  src/drone/CMakeLists.txt:7 (find_package)

all the other third party library dependencies are being found in the install path. help would be appreciated.
I already tried adding 
`include directorys(/work/build/third_party/MAVSDK/MAVSDK/src/MAVSDK-build/third_party/install)`
but that didn't change anything.

JonasVautherin, 2022-03-21:
Did you try using [`DEPS_INSTALL_PATH`](https://github.com/mavlink/MAVSDK/blob/main/CMakeLists.txt#L45)?

BestITUserEUW, 2022-03-22:
Thanks that solved the issue

closed 2022-03-22 by BestITUserEUW