mavlink/MAVSDK #1729: std::optional for CommandLong always returns true.

PeterQFR, 2022-04-05:
[Line 429](https://github.com/mavlink/MAVSDK/blob/4dcee2b14945a80276f1b30abd99088ed7009d9f/src/mavsdk/core/mavlink_command_sender.cpp#L440) of mavlink_commands.cpp,  is a check to see if particular parameter on the command long has been set using the contextural conversion of std::optional. If it has not been set, the the value is initialised to certain values depending on whether the auto pilot is ardupilot or not. 

It does appear that this check always returns true, regardless if the param has been set. This maybe due to the initialisation of the values in CommandLong to NAN. This will result in ardupilot autopilots throwing a SIGFPE and closing the connection. This may be the cause of some of the problems in #728 and elsewhere particularly when SITL is used. 

I'm not sure of the correct solution but perhaps not initialising those parameters is probably an option?

julianoes, 2022-04-06:
Aha, thanks for the note. Is this for all commands that are sent or just one specific one?

PeterQFR, 2022-04-06:
The Ardupilot failure comes from CommandLong. They try to convert it to a CommandInt for logging purposes and it is this that causes the fault. I comment out the conversion and I don't have problems. I also don't have commands logged though. 

Some of the values I have seen, particularly for param6 were uninitialised values e.g. 2.452.. e+38. In my code, I'm not initialising those values in the mavsdk::MavlinkPassthrough::CommandLong that I am using. 
This also faults during the systemImpl [sending of a command long](https://github.com/mavlink/MAVSDK/blob/4dcee2b14945a80276f1b30abd99088ed7009d9f/src/mavsdk/core/system_impl.cpp#L512) for autopilot capabilities.

I would have expected the check on Line429 to initialise those values to the correct ones, but it doesn't appear to work.
 The Values for command types are not std::optional in the MavlinkPassthrough[ definitions](https://github.com/mavlink/MAVSDK/blob/4dcee2b14945a80276f1b30abd99088ed7009d9f/src/mavsdk/plugins/mavlink_passthrough/include/plugins/mavlink_passthrough/mavlink_passthrough.h#L87) of CommandLong or CommandInt. In this case as params are not initialised as a std::optional its probably part of the reason the check in mavlink_commands.cpp fails.

PeterQFR, 2022-04-06:
Actually, maybe this is only an issue with the mavlink passthrough command long. I went back undid my changes to ardupilot and initialised those parameters. I haven't seen it fault on the std::optional CommandLong used in systemImpl, I think I got that wrong. 
So I'd say this is limited to mavlink_passthrough CommandLong, as they are not initialised as std::optional. This makes more sense.

julianoes, 2022-04-07:
Oh, I think this one is wrong actually:
https://github.com/mavlink/MAVSDK/blob/4dcee2b14945a80276f1b30abd99088ed7009d9f/src/mavsdk/core/system_impl.cpp#L515

It should be zero initialized.

PeterQFR, 2022-04-07:
what about the[ initialiser](https://github.com/mavlink/MAVSDK/blob/4dcee2b14945a80276f1b30abd99088ed7009d9f/src/mavsdk/plugins/mavlink_passthrough/include/plugins/mavlink_passthrough/mavlink_passthrough.h#L87) for mavlink_passthrough::Command

julianoes, 2022-04-07:
Interesting, so for passthrough it's different... Well, given it's "raw passthrough" it's just up to the user. I would say it's best if you zero-initialize the struct before setting the rest of the values. Does that make sense?

PeterQFR, 2022-04-07:
Well they're not raw types to start with, ie. we are not interfacing with mavlink passthrough using mavlink_msg_command_long_t.
Its a new type defined in mavlink_passthrough, so I don't think the "raw passthrough" applies. It also, as far as I can tell, goes through the same code path and calls maybe_reserved() on its way out. I think a solution would be to typedef the mavlink_passthrough::CommandLong  definition to the one in [Mavlink Command Sender ](https://github.com/mavlink/MAVSDK/blob/4dcee2b14945a80276f1b30abd99088ed7009d9f/src/mavsdk/core/mavlink_command_sender.h#L57).

Its up to you, but it would solve this issue for future, assuming that the command in mavlink passthrough uses the code path through the mavlink command sender.

julianoes, 2022-04-10:
You're right

julianoes, 2023-03-13:
I finally had a closer look at this. I can see where you're coming from. However, given it's a passthrough API, I would say that it's just up to the user to set all values to something that the flight stack can cope with. So, for PX4 that means filling the reserved values with NAN, and for ArduPilot it might have to be 0 in some (older AP versions?) cases as NAN will cause a fault. I'm reluctant to make this more complex than it is, unless you have a suggestion how this could easily be improved?

Closing, re-open if you still think this needs to be addressed and change.

PR #1731 (2022-04-07): core: fix uninitialized command
Fixes #1729.
closed 2023-03-13 by julianoes