mavlink/MAVSDK #1734: Cannot use QGroundControl simultaneously with an active mavsdk script in hardware testing

maximusron, 2022-04-08:
In hardware testing, QGroundControl accesses the serial port of my telemetry radio. This does not allow me to launch a MAVSDK script. My current modus operandi is to run the script without QGC but that is no longer viable. I have tried enabling port forwarding on QGC and changing the ```add_any_connection("serial:///``` to  ```add_udp_connection```.
This does result in the MAVSDK script connecting to the drone while QGC is open. But during mission upload, it fails as the ``` while (telemetry.health_all_ok()``` does not return true and the mission upload fails.
The output from the MAVSDK script is that it discovers many systems and components.
Is there any way to find out the correct system/component ID of the flight firmware?
Are there any alternative ways of enabling port forwarding/ keeping QGC open while being able to run a MAVSK script?
It happens smoothly in SITL where the MAVSDK script connects to the local PX4 flight firmware being launched on my system while QGC remains connected to it.

JonasVautherin, 2022-04-08:
I'm not completely sure what the forwarding in QGC does. I would try using mavlink-router:
* mavlink-router connects to the serial port
* mavlink-router sends mavlink to e.g. 14550 for QGC
* mavlink-router sends mavlink to e.g. 14540 for MAVSDK

Would that make sense?

julianoes, 2022-04-13:
You can also add connections with forwarding to MAVSDK: https://github.com/mavlink/MAVSDK/blob/0c13d0fea923ef5ef97657e70e7616ffaa309a5c/src/mavsdk/core/include/mavsdk/mavsdk.h#L89

maximusron, 2022-04-25:
> I'm not completely sure what the forwarding in QGC does. I would try using mavlink-router:
> 
> * mavlink-router connects to the serial port
> * mavlink-router sends mavlink to e.g. 14550 for QGC
> * mavlink-router sends mavlink to e.g. 14540 for MAVSDK
> 
> Would that make sense?

Yes mavlink router does solve the problem, thanks.
Although I could not figure out why Telemetry forwarding in QGC did not work, it is supposed to forward the radio on a local udp port. But that's an issue for QGC :). Thanks!!