mavlink/MAVSDK #1743: Implementing Missions for Ardupilot with MAVSDK

prokas-nikos, 2022-04-18:
When I export a mission from Mission planner I get the following output with mission commands.

QGC WPL 110
0	1	0	0	0	0	0	0	0	0	0	1
1	0	3	22	0.00000000	0.00000000	0.00000000	0.00000000	39.08962930	26.54831440	100.000000	1
2	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	39.08967920	26.54882670	100.000000	1
3	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	39.09013620	26.54875960	100.000000	1
4	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	39.09013310	26.54816690	100.000000	1
5	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	39.08976980	26.54822190	100.000000	1
6	0	3	21	0.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.000000	1

The first command of the above is not clear to me what it does.
I can right-click on SITL, import that file to ardupilot and I can make this flight in AUTO mode. 

I am trying to instead of uploading the mission through SITL to upload it from a custom android app. I have tried to implement this with Mission Plugin. As I have understood this is not working for ardupilot. 
I can see that @bazfp has already made this work with plugin MissionRaw. I have seen here [https://github.com/mavlink/MAVSDK/issues/1568](url) that "First mission item must be home position for ArduPilot". I cannot understand if this has to be a MAV_CMD_NAV_WAYPOINT instruction and then I have to add the MAV_CMD_NAV_TAKEOFF command?

If anyone has made this work, since there is no example in MAVSDK-Java nor MAVSDK-Python about mission_raw plugin, could you please provide me with the shortest possible example(seq 0, 1 and 2) in any language(C++, Java, Python) so that I can understand the process of deploying such commands?

julianoes, 2022-05-09:
The first command is supposedly the home position, or in this case it's just nothing, so not set as part of the mission.

What you need to do now is to parse the space separated values that you have above and map them to the mission_raw fields.

It looks like the format is described here:
https://mavlink.io/en/file_formats/#mission_plain_text_file

It would be nice to create an importer for this. Let's keep this issue open as a feature request.

prokas-nikos, 2022-05-23:
Thank you for the answer @julianoes.

The code(kotlin) which runs for me for now at least. I am planning to make some additions to it is the following.
I needed to have a way of picking up markers from maps layer and upload them as a flight plan. 

public fun uploadMission(waypoints : ArrayList<LatLng>, currentPos : Location)
    {
        var alt : Float = 40.0F

        var missionitems : ArrayList<MissionRaw.MissionItem> = ArrayList<MissionRaw.MissionItem>()
        var seq : Int = 0

        missionitems.add(
            MissionRaw.MissionItem(seq++,
                3, //MAV_FRAME_GLOBAL_RELATIVE_ALT
                22, //MAV_CMD_NAV_TAKEOFF
                1,
                1,
                0.0F,
                0.0F,
                0.0F,
                Float.NaN,
                0,
                0,
                alt,
                0)
        )

        missionitems.add(
            MissionRaw.MissionItem(seq++,
                3, //MAV_FRAME_GLOBAL_RELATIVE_ALT
                22, //MAV_CMD_NAV_TAKEOFF
                0,
                1,
                0.0F,
                0.0F,
                0.0F,
                Float.NaN,
                (currentPos.latitude * 10.0F.pow(7)).toInt(),
                (currentPos.longitude * 10.0F.pow(7)).toInt(),
                alt,
                0)
        )

        for( w in waypoints)
            missionitems.add(
                MissionRaw.MissionItem(seq++,
                    3, //MAV_FRAME_GLOBAL_RELATIVE_ALT
                    16, //MAV_CMD_NAV_WAYPOINT
                    0,
                    1,
                    0.0F,
                    0.0F,
                    0.0F,
                    Float.NaN,
                    (w.latitude * 10.0F.pow(7)).toInt(),
                    (w.longitude * 10.0F.pow(7)).toInt(),
                    alt,
                    0)
            )

        missionitems.add(
            MissionRaw.MissionItem(seq++,
                2,
                20, //MAV_CMD_NAV_RTL
                0,
                1,
                0.0F,
                0.0F,
                0.0F,
                0.0F,
                0,
                0,
                0.0F,
                0)
        )

        drone?.missionRaw?.uploadMission(missionitems)
            ?.doOnComplete() {
                    Log.i(Log.INFO.toString(),"uploadMission succeeded!")
                }
            ?.subscribe()
    }

I am going to make this importer in kotlin in the future because I want to be able to upload both qgroundcontrol and mission planner mission flights. I will try to make this mission planner importer in c++ and commit it.

julianoes, 2022-05-23:
Contributions are welcome :smiley:.

Here is the xml import for inspiration:
https://github.com/mavlink/MAVSDK/blob/bbdb9e96f5567a488925093f641d249f8e01b2de/src/mavsdk/plugins/mission_raw/mission_raw_impl.cpp#L474-L488

https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission_raw/mission_import.h
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission_raw/mission_import.cpp
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission_raw/mission_import_test.cpp

jankrystianserafim, 2022-08-16:
Hello Guys,

Have any of you managed to succesfully upload .plan files generated by QGC to a drone using MAVSDK?

PR #2601 (2025-07-07): mission_raw: enable MissionPlanner mission file parsing
This implement importing mission from the the old QGC txt export format apparently still used by MissionPlanner.

Closes #1743 

Depends on https://github.com/mavlink/MAVSDK-Proto/pull/378

According to https://mavlink.io/en/file_formats/#mission_plain_text_file

Given I just vibe coded this, I better review it myself tomorrow.
closed 2025-07-07 by julianoes