mavlink/MAVSDK #1744: Question: How can I get the value of channel 10 (1100 / 1900) ?

Madezz-o, 2022-04-19:
Hello ! 

I would like to get channel 10 value and I can't get it back with the "mavlink_msg_rc_channels_get_chan10_raw(&message);" function.

First, I tried in java then today in C++ so I would appreciate any help.
Today, my call to the function returns me the value : 65535, no matter if the pwm is 1100 or 1900.

This is my code : 

`     mavlink_message_t message;

        switch (message.msgid) {
            case MAVLINK_MSG_ID_RC_CHANNELS: {
                auto chan10_value = mavlink_msg_rc_channels_get_chan10_raw(&message);
                cout << chan10_value << "\n";
            } break;
            default:
                break;
        }`

Sorry for my English and thank you in advance !

julianoes, 2022-05-09:
I would check first using QGroundControl in mavlink inspector if the values are actually sent correctly.

Madezz-o, 2022-05-09:
> 

Of course this value change in QGroundControl...

julianoes, 2022-05-09:
Ok. Can you share the full example how this problem can be reproduced please? Then I can judge if actually something is wrong.

julianoes, 2024-05-27:
Silence...closing.