mavlink/MAVSDK #1749: Get current speed of the drone and Change heading when drone in hold mode

CY-1992, 2022-04-20:
Hi,

I would like to get the current speed of the drone as telemetry when it is doing a mission. Could not find it in telemetry API. Is the ground velocity the current speed of the drone?
And I would also like to change the current heading of the drone when it is in HOLD mode.  Is there anyway to do it ?

Thanks in advance.

JonasVautherin, 2022-04-20:
> Is the ground velocity the current speed of the drone?

I guess there are two speeds you may be interested in:
1. The [ground speed](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/telemetry/telemetry.proto#L38)
2. The airspeed (which comes with the [`FixedwingMetrics`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/telemetry/telemetry.proto#L755)).

Now if what's unclear is the meaning of "ground velocity":
* The ground speed is the speed relative to the ground. So if you are on the ground and the ground speed says "2 m/s", then for you the drone is moving at 2 m/s in some direction.
* The airspeed is the speed relative to the air. So if the drone flies against the wind at 2 m/s, and the wind speed is 2 m/s, then the ground speed will be 0, but the airspeed will be 2. The drone is flying at 2 m/s, but from your perspective on the ground, it is not "moving".

Does that help?

CY-1992, 2022-04-21:
@JonasVautherin. Yeah It helps. Thanks for quick response and explanation.  Camera is fixed on our drone, so I am trying to turn the entire drone to cover the area in the picture. During mission I am using regular yaw field in mission item. But when drone is put in to hold mode, How can I change the orientation of the drone at the current position. I am just thinking to use  [goto_location](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_action.html#classmavsdk_1_1_action_1afb3546fa994357e491816f2032716818) with current positional info with changes in yaw value. Is there any better way to achieve this. Any suggestions / help would be appreciated.

Thanks in advance

JonasVautherin, 2022-04-21:
I would use `goto_location`, I think :+1:

CY-1992, 2022-04-21:
@JonasVautherin Thanks. `goto_location` is working.

closed 2022-04-21 by CY-1992