mavlink/MAVSDK #1754: .deb file for Ubuntu 18.04 Jetson nano

Jai-GAY, 2022-04-27:
is there [such a package](https://github.com/mavlink/MAVSDK/releases)?

dpkg: error processing archive libmavsdk-dev_1.3.1_ubuntu18.04_amd64.deb (--install):
 package architecture (amd64) does not match system (arm64)
Errors were encountered while processing:
 libmavsdk-dev_1.3.1_ubuntu18.04_amd64.deb
 
Package: nvidia-jetpack
Version: 4.6.1-b110
Architecture: arm64
Maintainer: NVIDIA Corporation

JonasVautherin, 2022-04-27:
Did you try the [libmavsdk-dev_1.3.1_debian11_arm64.deb](https://github.com/mavlink/MAVSDK/releases/download/v1.3.1/libmavsdk-dev_1.3.1_debian11_arm64.deb) package?

Jai-GAY, 2022-04-28:
It seems can, but how do I [diagnose and debug](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues) it is working correctly?

Selecting previously unselected package libmavsdk-dev.
(Reading database ... 183367 files and directories currently installed.)
Preparing to unpack libmavsdk-dev_1.3.1_debian11_arm64.deb ...
Unpacking libmavsdk-dev (1.3.1) ...
Setting up libmavsdk-dev (1.3.1) ...
Processing triggers for libc-bin (2.27-3ubuntu1.5) ...

julianoes, 2022-05-09:
When you install the library, it installs the headers in `/usr/include/mavsdk` and the library file in `/usr/lib/libmavsdk.so.1`. To try it out you would try to compile your application, or an example from the examples directory. When you build an application it will include the headers and link to the library.

mholzel, 2023-07-19:
@julianoes I am having issues on a Jetson Orin with the default Debian 11 build. I can build and install it locally, but would like to generate a .deb that would make it easier to install on other machines I have. Is there documentation for how you are generating your .deb's? Or could you loosely outline the process for me?

julianoes, 2023-07-21:
We use [dockcross](https://github.com/dockcross/dockcross).

Here are some docs: https://mavsdk.mavlink.io/main/en/cpp/guide/build_mavsdk_server.html#cross_compilation_dockcross

And for the details on how each build is done, check out the GitHub CI: https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L267

closed 2024-05-27 by julianoes