mavlink/MAVSDK #1756: Follow Me messages getting sent in burst form (2 at once)

junwoo091400, 2022-04-28:
## Description
While developing the [improved Follow-Me](https://github.com/PX4/PX4-Autopilot/pull/19260), **it was spotted that 2 follow-me messages were getting received roughly 0.1 seconds apart. The expected behavior is a steady 1 Hz message received**, therefore it raised speculation that MAVSDK is sending 2 messages around the same time.

The Message Burst behavior is shown below, with a **vertical (infinite derivative) plot of data points in Follow-Me messages begin received.** Indicating that they were published around the same time by MAVSDK (burst-published)

[image]

## Ways to reproduce
1. Clone the Follow-Me MAVSDK script : https://github.com/junwoo091400/px4-follow-me-sim
2. Follow the instructions of README to install dependencies (which will install MAVSDK v1.2.0 in virtual environment for python)
3. Fetch and checkout the branch that I am developing on : https://github.com/junwoo091400/PX4-Autopilot/tree/followme_refactor
4. Run the SITL with the branch in step.3 : `make px4_sitl gazebo HEADLESS=1 `
5. Run the script in the px4-follow-me-sim repo from step.1 : `python3 follow-target-sim.py --model go_and_stop`
6. Wait for script to run ~ 30 seconds
7. Open the log of SITL with a PlotJuggler
8. Plot the 'follow_target' message logged. It will have vertical plot points like shown above.

## Few Notes
1. This happens both in using the MAVSDK v1.12.0 (Only a month old) and v0.22.0 (3 years old) - *image attached below
2. There was a fix for a similar (potentially the same) issue 1 year ago : #1405 
3. The burst-sending happens only from time to time (irregularly)

Here, the left side is a SITL result of using MAVSDK v1.12.0 and right is using v0.22.0:
[image]

julianoes, 2022-04-30:
Thanks for the detailed issue @junwoo091400. @potaito any idea?

potaito, 2022-05-02:
@julianoes not really :/

If I recall correctly, this was supposed to fix it: https://github.com/mavlink/MAVSDK/pull/1405

Back then we also speculated that sending messages at 1 Hz might cause the auto-resend at 1 Hz to trigger.