mavlink/MAVSDK #1757: Switching to Offboard is Currently not available

Eastonboy99, 2022-04-28:
Hello,

I have been running into an issue lately with enabling offboard mode. To my understanding, you need to set the velocity setpoint to something and MAVSDK runs a thread to send the setpoints at the correct frequency required by PX4. I do that and whenever I try to switch to offboard mode, I get Switching to Offboard is Currently not available. Sometimes it does work but it seems to be luck when it does.

Is there something I am doing wrong?

MAVSDK v0.50.1
PX4 1.12.3

JonasVautherin, 2022-04-28:
> Is there something I am doing wrong?

Difficult to say without seeing the code you are running :sweat_smile:. Can you share a small reproducible example that we could run, say, against SITL?

And before that, did you try running the offboard examples? Do they work?

Eastonboy99, 2022-04-29:
Here is a simple example. What I would like to do is switch to offboard mode from the transmitter. The example from MAVSDK sets offboard mode using offboard.start(). I would like to switch mid flight. 

```cpp
#include <signal.h>
#include <iostream>
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/offboard/offboard.h>
#include <chrono>
#include <thread>

int main(int argc, char const *argv[])
{

    mavsdk::Mavsdk mavsdk;
    std::shared_ptr<mavsdk::System> system;
    mavsdk::Offboard *offboard;
    mavsdk::ConnectionResult connection_result = mavsdk::ConnectionResult::SystemNotConnected;
    uint retryCount = 0;

    while (connection_result != mavsdk::ConnectionResult::Success && retryCount < 20)
    {
        // connection_result = mavsdk.add_any_connection("serial:///dev/ttyACM0"); //USB
        connection_result = mavsdk.add_any_connection("serial:///dev/ttyTHS1:500000");
        // connection_result = mavsdk.add_any_connection("udp://:14550");
        if (retryCount > 0)
        {
            std::this_thread::sleep_for(std::chrono::seconds(10));
        }
        retryCount++;
    }
    if (connection_result != mavsdk::ConnectionResult::Success)
    {
        throw std::runtime_error("PX4 Connection failed");
    }
    while (mavsdk.systems().size() == 0)
    {
        std::this_thread::sleep_for(std::chrono::seconds(1));
    }

    system = mavsdk.systems().at(0);
    if (!system)
    {
        throw std::runtime_error("System not found");
    }

    offboard = new mavsdk::Offboard(system);

    // sets velocity setpoint to 0.0 to prevent unwanted actions when enabling
    offboard->set_velocity_body({0.0f, 0.0f, 0.0f, 0.0f});
    // begin flight

    offboard->set_velocity_body({0.1f, 0.0f, 0.0f, 0.0f});
    while (true)
    {
    };

    return 0;
}
```

Eastonboy99, 2022-04-29:
But that puts it in offboard mode. I only want offboard mode to be enabled when I flip the switch on my remote that is set using QGroundControl.

JonasVautherin, 2022-04-29:
Oh, I see. So [here is](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/offboard/offboard_impl.cpp#L46) what `offboard->start();` does:

```c++
Offboard::Result OffboardImpl::start()
{
    {
        std::lock_guard<std::mutex> lock(_mutex);
        if (_mode == Mode::NotActive) {
            return Offboard::Result::NoSetpointSet;
        }
        _last_started = _time.steady_time();
    }

    return offboard_result_from_command_result(
        _parent->set_flight_mode(SystemImpl::FlightMode::Offboard));
}
```

I see 3 things being done there:

1. Check that the plugin is "active", in the sense that some setpoint has been set. You "activate" it when you call [`set_velocity_body`](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/offboard/offboard_impl.cpp#L283).
2. Set the `_last_started` value.
3. Set the flight mode to `Offboard`, which is what you do with your RC switch.

So let's look at what this `_last_started` [does](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/offboard/offboard_impl.cpp#L784):

```c++
        // Right after we started offboard we might still be getting heartbeats
        // from earlier which don't indicate offboard mode yet.
        // Therefore, we make sure we don't stop too eagerly and ignore
        // possibly stale heartbeats for some time.
        std::lock_guard<std::mutex> lock(_mutex);
        if (!offboard_mode_active && _mode != Mode::NotActive &&
            _time.elapsed_since_s(_last_started) > 3.0) {
            // It seems that we are no longer in offboard mode but still trying to send
            // setpoints. Let's stop for now.
            stop_sending_setpoints();
        }
```

My understanding here is that if we are not in `Offboard` mode anymore, then MAVSDK calls `stop_sending_setpoints();` (after 3 seconds since the last `start` call) :thinking:. And `stop_sending_setpoints` moves it back to "not active".

So... short answer is that what you want to do is not supported: MAVSDK currently _wants_ `offboard->start()` to be called. More importantly, MAVSDK needs a way to detect when it should stop sending the setpoints.

I am not completely sure how to solve that: how can MAVSDK know that you plan on starting the offboard mode later from a different channel? Or should MAVSDK always send the setpoints when the autopilot is in `Offboard` mode unless `offboard->stop()` has been called explicitly? Maybe that could make sense indeed :thinking:.

@julianoes would you have an opinion on that? I guess mostly in terms of safety (we don't want MAVSDK to do weird stuff with offboard mode and cause a crash, obviously).

julianoes, 2022-05-09:
There have been similar requests in the past: #1307. It's on my todo list which is getting longer and longer, sorry about that.