mavlink/MAVSDK #1769: Switch CI to Ubuntu 22.04

julianoes, 2022-05-09:
Now that the next Ubuntu LTS release is out, we probably want to move the main support from Ubuntu 20.04 to Ubuntu 22.04. We will keep maintenance for Ubuntu 20.04 as long as it's easily feasbile but it's probably soon time to sunset support for 18.04, as its cmake version is outdated anyway.

reedev, 2022-05-31:
Please keep 18.04 too for a while since JetPack for NVIDIA Jetson is still 18.04. Of course, when the requirement for a more recent cmake for instance is needed, that would be fine imho

JonasVautherin, 2022-05-31:
Don't we already require a cmake newer than what 18.04 provides? :thinking: 

Anyway I agree with @reedev, IMO we should keep 18.04 until it is end of life (that's 23.04, right?) unless it becomes super tricky to maintain.

JonasVautherin, 2022-05-31:
Anyway we are usually trying hard to support the Ubuntu LTS, just for cmake we needed the newer cmake at some point, I don't recall why. But e.g. even for tinyxml2 we patch them because their policy is that people should just install the latest cmake, and at least I disagree with it :innocent: : https://github.com/mavlink/MAVSDK/blob/main/third_party/tinyxml2/cmake-3.10.2.patch

reedev, 2022-05-31:
that's a rather interesting read, thanks

julianoes, 2022-05-31:
> See the discussion https://github.com/leethomason/tinyxml2/pull/866 if you are interested.

Ha, strong opinions indeed.

JonasVautherin, 2022-06-01:
https://github.com/mavlink/MAVSDK/pull/1797 goes in that direction.

julianoes, 2022-06-02:
Let me add it to CI.

julianoes, 2024-05-27:
I think I did this a while ago.

closed 2024-05-27 by julianoes