mavlink/MAVSDK #1778: Unsupported Mission Commands from QGroundcontrol created mission

ankitatcodingnebula, 2022-05-19:
Hi MAVSDK Team,

Any mission created from QGroundcontrol throws "UNSUPPORTED mission item" error for few of the MAVLink Commands (MAV_CMD). 
QGC mission creation adds the following MAV_CMD commands to the mission plan by default unless unchecked

1. MAV_CMD_NAV_TAKEOFF ([22](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_TAKEOFF) )
2. MAV_CMD_SET_CAMERA_MODE ([530](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_MODE) )

When Download Mission api is called MAVSDK throws error which can be seen on mavsdk server console

```

[04:36:45|Debug] Assembling Message: 0 (mission_impl.cpp:612)
[04:36:45|Error] UNSUPPORTED mission item command (530) (mission_impl.cpp:743)
```

Looking at `DownloadMissionCallback` of `mission_impl.cpp` these commands are not part of the if else condition. Could you please confirm if there is a reason these have been left out? Is there any workaround to fix the UNSUPPORTED error on download mission?

JonasVautherin, 2022-05-19:
I guess they should be added to the mavsdk mission plugin? Would you be willing to contribute that? :blush:

julianoes, 2022-05-19:
@ankitatcodingnebula thanks for the note.

First off: the mission plugin only supports a subset of commands. If you need all possible commands/mission items, you need to use the `MissionRaw` plugin instead.

Now, for your specific commands, I can see how it would be nice to have support for them:
1. MAV_CMD_NAV_TAKEOFF: there is work in progress to add support for this, see: #1628.
2. MAV_CMD_SET_CAMERA_ACTION: given we support taking photos and videos in CameraAction it would seem logical that we also implement/support switching between photo and video mode. It would be nice to add that capability. The first step would be to add it here: https://github.com/mavlink/MAVSDK-Proto/blob/0c4379fb476b41f5193de7cd602e24668c0fb268/protos/mission/mission.proto#L222.

julianoes, 2025-06-26:
The QGC import is now just done in MissionRaw, so we don't have all the compatibility.
Closing.

closed 2025-06-26 by julianoes