mavlink/MAVSDK #1785: get_all_params error (std::future_error: Promise already satisfied)

elbarto1980, 2022-05-21:
Hi, I am running MAVSDK 1.4.0 on Ubuntu 20.04.
I am trying to get all PX4 params using the get_all_params() function and I noticed that the `prom.set_value(all_params);` line of code causes a segmentation fault. It happens every time, if the program does not terminate immediately after receiving the parameters list.
The error I get is:
> terminate called after throwing an instance of 'std::future_error'
>   what():  std::future_error: Promise already satisfied

I think this issue might be somehow related to another issue ( #1527) that @julianoes fixed on September 2021.

Hopefully, the few following lines of code should be useful to reproduce the error:

```
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/param/param.h>
#include <thread>
#include <iostream>

using namespace mavsdk;

int main() {
    Mavsdk mavsdk;

    mavsdk.add_udp_connection();
    std::this_thread::sleep_for(std::chrono::seconds(2));

    auto system = mavsdk.systems().at(0);
    auto param = Param{system};
    auto all_params = param.get_all_params();

    for (auto param : all_params.int_params) {
        std::cout << param.name << std::endl;
    }

    int i = 60;
    do {
        std::this_thread::sleep_for(std::chrono::seconds(1));
        i--;
        std::cout << i << std::endl;
    } while (i>0);
}
```

Thank you!

julianoes, 2022-05-22:
Thanks for the issue @elbarto1980, and sorry that I broke that during the recent refactoring. I will look into it.

elbarto1980, 2022-05-23:
Thank you @julianoes for providing a fix so quickly!

julianoes, 2022-05-23:
Oh, a new twist on this? :thinking: Let's see.

julianoes, 2022-05-23:
There was indeed another bug, thanks for the nice example to reproduce it.

See: https://github.com/mavlink/MAVSDK/pull/1788/commits/5ce3a5c2c821e5273c09a56ecf719eb5b36d52df

closed 2022-05-24 by JonasVautherin

julianoes, 2022-05-26:
Thanks again for your help @elbarto1980. The minimal examples to reproduce this were very helpful, and allowed me to quickly pinpoint what was wrong!

julianoes, 2022-05-26:
Nice, yes I backported the fixes to [v1.4.2](https://github.com/mavlink/MAVSDK/releases/tag/v1.4.2), until v2.0.0 ships.

PR #1788 (2022-05-24): params: fix double promise call
This fixes a segfault/abort when using get_all_params(). The issue was that the timeout handler was not removed as it should have been, and so after a successful completion, we got another call from the timeout handler which then satisfies the promise a second time which is not possible.

This did not show up in the integration test (or at least not often) because the test was already destructed by the time the timeout would trigger.

Fixes #1785.