mavlink/MAVSDK #1786: Errors while using mavsdk::Geofence reference

maximusron, 2022-05-21:
I am trying to upload a geofence in my mavsdk script. Im getting errors while building. i suspect it has something to do with the constructor itself as other parts of the code like the structures itself do not give any errors. 
```bash
catkin_ws/devel/.private/link_uav/lib/liblink_uav.so: undefined reference to `mavsdk::Geofence::upload_geofence(std::vector<mavsdk::Geofence::Polygon, std::allocator<mavsdk::Geofence::Polygon> >) const'

catkin_ws/devel/.private/link_uav/lib/liblink_uav.so:  undefined reference to `mavsdk::Geofence::~Geofence()'

catkin_ws/devel/.private/link_uav/lib/liblink_uav.so:  undefined reference to `mavsdk::Geofence::Geofence(std::shared_ptr<mavsdk::System>)'

catkin_ws/devel/.private/link_uav/lib/liblink_uav.so:  undefined reference to `mavsdk::operator<<(std::ostream&, mavsdk::Geofence::Result const&)'
collect2: error: ld returned 1 exit status
```

Here is a snippet of the function I am using:

```bash
auto geofence = mavsdk::Geofence{system_};
  while(!telemetry_->health_all_ok())
  {
    std::cout<<"On standby for system to be ready."<<std::endl;
    std::this_thread::sleep_for(std::chrono::seconds(1));
  }

  std::cout<<"System Ready"<<std::endl;
  std::cout << "Creating and uploading geofence\n";

    std::vector<mavsdk::Geofence::Point> points;
    points.emplace_back(add_point(47.39929240, 8.54296524));
    points.emplace_back(add_point(47.39696482, 8.54161340));
    points.emplace_back(add_point(47.39626761, 8.54527193));
    points.emplace_back(add_point(47.39980072, 8.54736050));

    std::vector<mavsdk::Geofence::Polygon> polygons;
    mavsdk::Geofence::Polygon new_polygon{};
    new_polygon.fence_type = mavsdk::Geofence::Polygon::FenceType::Inclusion;
    new_polygon.points = points;

    polygons.push_back(new_polygon);

    {
        std::cout << "Uploading geofence...\n";

        const mavsdk::Geofence::Result result = geofence.upload_geofence(polygons);

        if (result != mavsdk::Geofence::Result::Success) {
            std::cerr << "Geofence upload failed: " << result << ", exiting.\n";
            return 1;
        }
        std::cout << "Geofence uploaded.\n";
    }
```
The mavsdk::System, Action and Telemetry constructors are initialised properly and the mavsdk::System object is named ```system_```

julianoes, 2022-05-22:
@maximusron which version of mavsdk do you have installed?

maximusron, 2022-05-23:
> @maximusron which version of mavsdk do you have installed?

I am currently using mavsdk 0.37, the development of the mavsdk scripts in our software had started quite a while back. The code I'm using for the geofence upload is from the examples code https://github.com/mavlink/MAVSDK/tree/develop/examples/geofence_inclusion
found on the mavsdk documentation for version 0.37.

julianoes, 2022-05-23:
With that old version you still need to link to each plugin:

```
target_link_libraries(your_binary
    MAVSDK::mavsdk_geofence
    MAVSDK::mavsdk
)
```

I assume you're missing `MAVSDK::mavsdk_geofence`?

julianoes, 2022-05-23:
With v1.0 this changed: https://mavsdk.mavlink.io/main/en/cpp/api_changes.html#10

maximusron, 2022-08-14:
Yes, that seemed to be the problem, thank you

closed 2022-08-14 by maximusron