mavlink/MAVSDK #1789: Command denied in logs when mission plugin is instantiated

CY-1992, 2022-05-23:
Hi,

When I just try to initialize mission plugin, in the logs there is one warning "command denied (512). (mavlink_command_sender.cpp:191)". I used the code below with jMAVSim SITL. 
```cpp
#include <chrono>
#include <cstdint>
#include <iostream>
#include <future>
#include <memory>
#include <thread>
#include <iostream>
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/action/action.h>
#include <mavsdk/plugins/mission/mission.h>

using namespace mavsdk;
using std::chrono::seconds;
using std::this_thread::sleep_for;

std::shared_ptr<System> get_system(Mavsdk& mavsdk)
{
    std::cout << "Waiting to discover system...\n";
    auto prom = std::promise<std::shared_ptr<System>>{};
    auto fut = prom.get_future();

    // We wait for new systems to be discovered, once we find one that has an
    // autopilot, we decide to use it.
    mavsdk.subscribe_on_new_system([&mavsdk, &prom]() {
        auto system = mavsdk.systems().back();

        if (system->has_autopilot()) {
            std::cout << "Discovered autopilot\n";

            // Unsubscribe again as we only want to find one system.
            mavsdk.subscribe_on_new_system(nullptr);
            prom.set_value(system);
        }
    });

    // We usually receive heartbeats at 1Hz, therefore we should find a
    // system after around 3 seconds max, surely.
    if (fut.wait_for(seconds(3)) == std::future_status::timeout) {
        std::cerr << "No autopilot found.\n";
        return {};
    }

    // Get discovered system now.
    return fut.get();
}

int main(int argc, char** argv)
{
    const std::string conn = "udp://:14540";
    Mavsdk mavsdk;
    ConnectionResult connection_result = mavsdk.add_any_connection(conn);
    auto system = get_system(mavsdk);
    if (!system) {
        return 1;
    }

    auto mission = Mission{system};

    sleep_for(seconds(10));
    std::cout << "Finished...\n";
    return 0;
}
```
and logs
[image]

Mission plugin works perfectly though. I am just curious to know why this command denied appears in the logs only with Mission plugin but not with any other plugins like Action or Telemetry. 

Thanks

CY-1992, 2022-05-23:
@JonasVautherin  Tried with `main` branch. 

```
ExternalProject_add(
    MAVSDK
    GIT_REPOSITORY https://github.com/mavlink/MAVSDK.git
    PREFIX MAVSDK
    GIT_TAG main 
    CMAKE_ARGS "${CMAKE_ARGS}"
)
```
And the result is same.

[image]

JonasVautherin, 2022-05-23:
Interesting... so there is some component on the system that answers "DENIED" to this `REQUEST_MESSAGE`, apparently? :thinking:

julianoes, 2022-05-23:
It might be requests to check for gimbal v2. I want to make these silent because they are not really an issue as it falls back to v1.

closed 2022-06-27 by CY-1992