mavlink/MAVSDK #1791: Build issue with zlib 1.2.12

hamishwillee, 2022-05-26:
Building MAVSDK from source following standard linux instructions here: https://mavsdk.mavlink.io/main/en/cpp/guide/build.html#release

```
sudo apt-get update
sudo apt-get install build-essential cmake git
git clone https://github.com/mavlink/MAVSDK.git
cd MAVSDK
git submodule update --init --recursive
cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
```
Platform is Ubuntu 20.0.4 (running in WSL, but that shouldn't matter).
The error is:
```
...
[100%] Built target tinyxml2
-- Preparing external project "zlib" with args:
-- -- -DCMAKE_INSTALL_PREFIX:PATH=/home/ubuntu/github/MAVSDK/build/default/third_party/install
-- -- -DCMAKE_TOOLCHAIN_FILE:PATH=
-- -- -DCMAKE_POSITION_INDEPENDENT_CODE=ON
-- -- -DCMAKE_BUILD_TYPE=Debug
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/github/MAVSDK/build/default/third_party/zlib
[ 12%] Performing update step for 'zlib'
error: Your local changes to the following files would be overwritten by checkout:
        CMakeLists.txt
Please commit your changes or stash them before you switch branches.
Aborting
CMake Error at /home/ubuntu/github/MAVSDK/build/default/third_party/zlib/zlib/tmp/zlib-gitupdate.cmake:144 (message):
  Failed to checkout tag: 'v1.2.12'
```

So that tag exists: https://github.com/madler/zlib/releases/tag/v1.2.12 and I can see this was updated in  https://github.com/mavlink/MAVSDK/commit/13166c0096d3f286466dff94b07e355bf6ee64f3

I tried reverting just that cmake back to 1.2.11 but got similar error with curl building
```
[ 25%] No patch step for 'curl'
From https://github.com/curl/curl
```

hamishwillee, 2022-05-26:
Ignore that. Works on Ubuntu. Need to work out what is wrong with my WSL.

closed 2022-05-26 by hamishwillee

julianoes, 2022-05-26:
I had that too today. It is because the dependencies have been updated. You need to do a fresh build by removing or renaming the build folder.