mavlink/MAVSDK #1800: [Rtk] Data above 180 bytes length not working

Katawann, 2022-06-06:
Hi,

I tested the new RTK plugin and it works like a charm ! Thank you @julianoes for the job.

In my working setup I have something like this:
Ntrip caster <---> Internet <---> Linux Ubuntu 20.04 <---> C++ app with MAVSDK and Ntrip Client <---> USB <---> Pixhawk 4

I noticed that if I send a data with length of 4*180 bytes (accepted by the plugin) it will never turn to RTK position, but sending 180 bytes works fine. Not sure to understand the logic around the fragment ID. Maybe the plugin should only accept 180 bytes maximum or is there a logic I missed ?

Katawann, 2022-06-07:
From my understanding, MAVProxy ntrip module is sending N packets of 180 bytes according to the size of the RTCM data [https://github.com/ArduPilot/MAVProxy/blob/65acf3a9cef818ea819ac0ccac3c5a9577bd1d06/MAVProxy/modules/mavproxy_ntrip.py#L115](https://github.com/ArduPilot/MAVProxy/blob/65acf3a9cef818ea819ac0ccac3c5a9577bd1d06/MAVProxy/modules/mavproxy_ntrip.py#L115)

JonasVautherin, 2022-06-07:
Oh, I misread the code, sorry :see_no_evil:. The specs read:

> RTCM message (may be fragmented)

And the code is apparently trying to split the data in multiple messages. So that's something to debug indeed :+1:.

julianoes, 2022-06-18:
Oops, that's indeed wrong.

closed 2022-06-20 by julianoes

Katawann, 2022-06-20:
Awesome thank you good job. I'll test it and come back here if I still see an issue