mavlink/MAVSDK #1803: Use 2 UDP ports for sending and receiving data

Consti10, 2022-06-10:
Hello,

I am wondering if it is possible to setup mavsdk with a UDP connection where messages are received on one localhost::udp tuple and sent to another localhost::udp tuple.

Quick explainer: the method we used to go with in OpenHD has always been: 
On the ground unit,
1) listen for messages from the Ground controll application on one udp localhost port
2) send messages to the ground controll application on another udp localhost port.

So for example, the ground controll application listens on localhost::16550 and sends data to localhost::16551

However, I cannot see a way to configure mavsdk to listen and send to different udp ports.
Since the send udp port is automatically added by using the address and port of a received packet.

Best regards, 
Constantin

JonasVautherin, 2022-06-10:
Sounds like you have a proxy that sends to localhost:16550 and receives on 16551, right? And I'm guessing your proxy then sends to/receives from the drone with wifi-broadcast? I really don't know the setup, I just read about wifi-broadcast a while ago :innocent:.

But if it is like this, whenever MAVSDK receives mavlink packets from your proxy (which in turn receives them from the drone), it will discover it as a drone, and it will start answering to it. I don't think you need to change anything in MAVSDK :thinking:.

What am I missing there?

Consti10, 2022-06-10:
Yes, that is correct.

The problem is that mavsdk receives packets from localhost::16551 and then discovers the drone (nice ;) ) but since the communication port back is automated (it is set to the address::port tuple it gets from the received udp packets) it then sends data back via localhost::16551 instead of localhost::16550

Maybe we also have a worm somewhere how we did the mavlink forwarding all the time - but if the proxy (so to say) and the ground controll with mavsk are both running on the same system, rn the communication can only be in one direction.

If the proxy had its own ip then it would of course be no problem - as an example, how I think it is intended from mavsdk:
Proxy: has IP 192.168.0.11 and listens on 14550
And sends to 0.0.0.0:14550
And mavsdk in turn then:
Listens on 0.0.0.0:14550 - and as soon as it gets a udp packet it knows it can send to 192.168.0.11::14550 for communication back.

Consti10, 2022-06-10:
For a more detailed explanation: The routing setup is simple:

(Flight Controll,own mavlink sys id) <-> (OpenHD Air, own sys id) <-> (OpenHD Ground, own sys id)

So via the OpenHD ground unit, one can send and receive mavlink messages to FC, Air and Ground by mavlink sys id.

And the stream QOpenHD / QGroundControll gets from the ground unit (in our sense, it only makes sense to connect the ground controll application to the ground unit, either running directly on it or with an external device, like phone) therefore contains multiplexed mavlink messages from FC, OpenHD Air and OpenHD Ground.

Discovery of the systems with mavsdk works flawlessly (mavlink in). How to communicate back I am still unsure.

Consti10, 2022-06-10:
any advice ?

Consti10, 2022-06-10:
Yeah as I said, it is possible that we do something stupid there.

But in a nutshell:

rn I'd need a function like this in mavsdk:

```
// expects incoming messages at listenIp::listenPort
// and sends messages out to sendIp::sendPort
ConnectionResult add_udp_connection(
        std::string listenIp,
        int listenPort,
        std::string sendIp,
        int sendPort)
```
        
but mavsdk as a "feature" wants to free me from the hustle of manually specifying the sendIp::sendPort :
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/udp_connection.cpp#L235

Which r.n. I cannot use.

JonasVautherin, 2022-06-10:
Yeah it does not support it at the moment, so you'd have to implement that feature.

The thing is that it seems very weird to me, so I was trying to understand to see if there is not a way to make it work like it's intended :sweat_smile:.

Out of curiosity, how do you connect QGC? Did you modify QGC such that it uses two different UDP sockets to receive and send?

Consti10, 2022-06-10:
> Yeah it does not support it at the moment, so you'd have to implement that feature.
> 
> The thing is that it seems very weird to me, so I was trying to understand to see if there is not a way to make it work like it's intended sweat_smile.

jup, I understand that. R.n i am trying to figure out how to change OpenHD in this regard, not mavsdk ;) 
Especially since the following would kinda be the same, and works:
1) Run the PX4 FC emulator on machine 1
2) start mavsdk on machine 1 (the same machine).
And then the FC emulator and mavsdk communicate on the same UDP port, but with different send/receive IP addresses, right  ?

> 
> Out of curiosity, how do you connect QGC? Did you modify QGC such that it uses two different UDP sockets to receive and send?

It is a while,  i think i never checked if sending messages from QGroundControll works, only if messages show up in QGroundControll console.

Consti10, 2022-06-10:
> > start mavsdk on machine 1 (the same machine).
> > And then the FC emulator and mavsdk communicate on the same UDP port, but with different send/receive IP addresses, right ?
> 
> Well if they run on the same machine, they should rather have the same IP (localhost), but use different ports 😅

Well, that is exactly the thing - how do the examples work then (the mavsdk examples that use px4 emulator) ;)

JonasVautherin, 2022-06-10:
Say you run SITL on your computer. SITL sends UDP to localhost:14540 from some ephemeral port. MAVSDK listens on 14540, and therefore receives the messages coming from SITL. When it does, then it takes the ephemeral port it sees in those messages, and uses it to send messages back to SITL.

Consti10, 2022-06-11:
So I managed to change OpenHD in this regard.
No idea why I had such a hard time wrapping my head around, but now it works the following:
OpenHD binds port 127.0.0.1:14551 and listens on 127.0.0.1:14550
AND
instead of using sendto() with a unbound port (which then in turn means the sender port can be anything) messages are sent with sendto() from the bound port (the same that is used for listening).

So messages from OpenHD to mavsdk go the following: 
OpenHD (out) via 127:0:0:1:14551 sent to 127:0:0:0:1:14550 

So when mavsdk receives the first message, the sender address::port is 127:0:0:1:14551 and mavsdk can send the messages back to 127:0:0:1:14551.

JonasVautherin, 2022-06-11:
Yeah that seems better. So it works now?

Consti10, 2022-06-13:
Yes, this issue can be closed. 
Here is some examle code:
https://github.com/Consti10/wifibroadcast/blob/f1769607769e4b3febc4d0966123da5706494adf/src/HelperSources/SocketHelper.hpp#L258

closed 2022-06-13 by JonasVautherin

Consti10, 2022-06-13:
A feature where one can add a customized connection (aka a class that just provides the apropriate callbacks) would be nice though.
Dummy code:

```
class CustomConnection : public mavsdk::Connection{
// has a custom send and receive imlementation
}

mavsdk::add_custom_connection(Connection customConnection)
```

In general, I am a bit divided r.n how much benifit mavsdk provides to us in its "public api" form. 
Since it is really easy to run into things where the public api hides away things that we'd need.

As an example, mavsdk does ping messages but does not expose the result of these ping messages via the public api. Of course, one could add the proper public api code, but that is not the only thing I already ran into.

JonasVautherin, 2022-06-13:
> A feature where one can add a customized connection

I maintain that using one socket to receive and one socket to send seems a bit weird to me, I have never seen UDP used like that :sweat_smile:. Now you're right that MAVSDK could have some kind of `add_connection(Connection connection)` interface, so that you could use whatever you want. Happy to review a PR, if you feel like doing that :blush:.

> In general, I am a bit divided r.n how much benifit mavsdk provides to us in its "public api" form.

To be fair, in C++, you have the `MavlinkPassthrough` module that allows you to do raw mavlink. It cannot really get lower level than that :sweat_smile:.