mavlink/MAVSDK #1823: Autopilot Server Example segfaults

kadrim, 2022-06-28:
Hi,

i am trying to learn with the MavSDK to implement Mavlink for a custom drone and wanted to rund and understand the `autopilot_server` example.

No matter which release (tested: 1.4.3, 1.0.0 and 0.44.0) on different Platforms (Rasberry Pi OS, Ubuntu Baremetal, official Docker-Images, WSL2 etc.), the example always segfaults.

I precompile the MAVSDK via:

```
cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
cmake --build build/default -j8
sudo cmake --build build/default --target install
```

which works flawlessly.

then, i build the example:

```
cd examples/autopilot_server
cmake -Bbuild -H.
cmake --build build -j4
```

Then i can run it via:

```
sudo build/autopilot_server
```

The output is as follows:

```
[05:14:43|Info ] MAVSDK version: v1.4.3 (mavsdk_impl.cpp:[05:14:43|Info ] MAVSDK version: v1.4.3 (mavsdk_impl.cpp:20)20)

Connected!
[05:14:43|Debug] Initializing connection to remote system... (mavsdk_impl.cpp:480)
Connected autopilot server side!
Sleeping AP thread...
[05:14:43|Info ] New system on: 127.0.0.1:42018 (with sysid: 1) (udp_connection.cpp:192)
[05:14:43|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:482)
[05:14:43|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[05:14:43|Debug] New: System ID: 245 Comp ID: 190 (mavsdk_impl.cpp:212)
[05:14:43|Debug] Component Unsupported component (190) added. (system_impl.cpp:377)
[05:14:43|Debug] Discovered 1 component(s) (system_impl.cpp:578)
Discovered Autopilot from Client
[05:14:44|Warn ] sending again after 0.506557 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)
[05:14:44|Warn ] sending again after 1.01474 s, retries to do: 2  (520). (mavlink_command_sender.cpp:287)
[05:14:45|Warn ] sending again after 1.52227 s, retries to do: 1  (520). (mavlink_command_sender.cpp:287)
[05:14:45|Error] Retrying failed (520) (mavlink_command_sender.cpp:307)
Setting System
MissionRawServer created
[05:14:53|Debug] Setting interval for msg id: 245 interval_ms:1000 (telemetry_server_impl.cpp:40)
[05:14:53|Debug] Request Param CAL_GYRO0_ID (mavlink_parameters.cpp:1442)
[05:14:53|Debug] Request Param CAL_ACC0_ID (mavlink_parameters.cpp:1442)
[05:14:53|Debug] Request Param CAL_MAG0_ID (mavlink_parameters.cpp:1442)
[05:14:53|Debug] Request Param SYS_HITL (mavlink_parameters.cpp:1442)
[05:14:59|Warn ] sending again after 0.508828 s, retries to do: 3  (512). (mavlink_command_sender.cpp:287)
[05:14:59|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:124)
[05:14:59|Warn ] sending again after 1.01691 s, retries to do: 2  (512). (mavlink_command_sender.cpp:287)
[05:15:00|Warn ] sending again after 1.52547 s, retries to do: 1  (512). (mavlink_command_sender.cpp:287)
[05:15:00|Error] Retrying failed (512) (mavlink_command_sender.cpp:307)
[05:15:03|Debug] Request Param my_param (mavlink_parameters.cpp:1442)
Found Param my_param: 1
Creating Mission
Uploading mission...
SystemID 
[05:15:03|Debug] Receive Mission Count in Server (mission_raw_server_impl.cpp:151)
Mission uploaded.
Received Uploaded Mission!
[... 220 lines cut ...]
    at /home/david/repos/MAVSDK/src/mavsdk/plugins/mission_raw_server/mission_raw_server_impl.cpp:183
#10 0x00007ffff7cc8266 in std::__invoke_impl<void, mavsdk::MissionRawServerImpl::init()::<lambda(const mavlink_message_t&)>::<lambda()>::<lambda(mavsdk::MAVLinkMissionTransfer::Result, std::vector<mavsdk::MAVLinkMissionTransfer::ItemInt>)>::<lambda()>&>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/11/bits/invoke.h:61
#11 0x00007ffff7cc7676 in std::__invoke_r<void, mavsdk::MissionRawServerImpl::init()::<lambda(const mavlink_message_t&)>::<lambda()>::<lambda(mavsdk::MAVLinkMissionTransfer::Result, std::vector<mavsdk::MAVLinkMissionTransfer::ItemInt>)>::<lambda()>&>(struct {...} &) (__fn=...) at /usr/include/c++/11/bits/invoke.h:111
#12 0x00007ffff7cc6c61 in std::_Function_handler<void(), mavsdk::MissionRawServerImpl::init()::<lambda(const mavlink_message_t&)>::<lambda()>::<lambda(mavsdk::MAVLinkMissionTransfer::Result, std::vector<mavsdk::MAVLinkMissionTransfer::ItemInt>)>::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/include/c++/11/bits/std_function.h:290
#13 0x00007ffff7b0ac7a in std::function<void ()>::operator()() const (this=0x7ffff5991d00)
    at /usr/include/c++/11/bits/std_function.h:590
#14 0x00007ffff7b39da5 in mavsdk::MavsdkImpl::process_user_callbacks_thread (this=0x7ffff0000b80)
    at /home/david/repos/MAVSDK/src/mavsdk/core/mavsdk_impl.cpp:591
#15 0x00007ffff7b46e7e in std::__invoke_impl<void, void (mavsdk::MavsdkImpl::*)(), mavsdk::MavsdkImpl*> (
    __f=@0x7ffff0001410: (void (mavsdk::MavsdkImpl::*)(mavsdk::MavsdkImpl * const)) 0x7ffff7b39c96 <mavsdk::MavsdkImpl::process_user_callbacks_thread()>, __t=@0x7ffff0001408: 0x7ffff0000b80) at /usr/include/c++/11/bits/invoke.h:74
#16 0x00007ffff7b46dd1 in std::__invoke<void (mavsdk::MavsdkImpl::*)(), mavsdk::MavsdkImpl*> (
    __fn=@0x7ffff0001410: (void (mavsdk::MavsdkImpl::*)(mavsdk::MavsdkImpl * const)) 0x7ffff7b39c96 <mavsdk::MavsdkImpl::process_user_callbacks_thread()>) at /usr/include/c++/11/bits/invoke.h:96
#17 0x00007ffff7b46d31 in std::thread::_Invoker<std::tuple<void (mavsdk::MavsdkImpl::*)(), mavsdk::MavsdkImpl*> >::_M_invoke<0ul, 1ul> (this=0x7ffff0001408) at /usr/include/c++/11/bits/std_thread.h:253
#18 0x00007ffff7b46bfc in std::thread::_Invoker<std::tuple<void (mavsdk::MavsdkImpl::*)(), mavsdk::MavsdkImpl*> >::operator() (this=0x7ffff0001408) at /usr/include/c++/11/bits/std_thread.h:260
#19 0x00007ffff7b46abc in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (mavsdk::MavsdkImpl::*)(), mavsdk::MavsdkImpl*> > >::_M_run (this=0x7ffff0001400) at /usr/include/c++/11/bits/std_thread.h:211
#20 0x00007ffff75a92c3 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#21 0x00007ffff7319b43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#22 0x00007ffff73aba00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

only the current master shows a somewhat different behaviour, as it doesn't segfault, but the vehicle never takes off.

Can someone point me in the right direction?

Thanks in advance!

julianoes, 2022-06-30:
That doesn't look so good. Thanks for the detailed bug report. I'll try to have a closer look.

julianoes, 2022-06-30:
> only the current master shows a somewhat different behaviour, as it doesn't segfault, but the vehicle never takes off.

Ah, so that means I probably fixed the segfault recently in https://github.com/mavlink/MAVSDK/pull/1733/commits/0240c2de4fc80c141ccf1cddb2462f2ce169de9b.

What's the output when it never takes off? Maybe something like "manual control lost"? In that case try this:

```
param->set_param_int("COM_RCL_EXCEPT", 7)
```

kadrim, 2022-06-30:
i just checked out the current master at df2bad28d7ab0b147118261762fd7325cdbc3d3c

the output is as follows:

```
sudo ./autopilot_server 
[05:04:35|Info ] MAVSDK version: v1.4.0-193-gdf2bad28 (mavsdk_impl.cpp:24)
Connected!
[05:04:35|Info ] MAVSDK version: v1.4.0-193-gdf2bad28 (mavsdk_impl.cpp:24)
[05:04:35|Debug] Initializing connection to remote system... (mavsdk_impl.cpp:602)
Connected autopilot server side!
[05:04:35|Info ] New system on: 127.0.0.1:34710 (with system ID: 1) (udp_connection.cpp:192)
MissionRawServer created
[05:04:35|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:604)
[05:04:35|Debug] Component Autopilot (1) added. (system_impl.cpp:382)
[05:04:35|Debug] New: System ID: 245 Comp ID: 190 (mavsdk_impl.cpp:309)
[05:04:35|Debug] Component Ground station (190) added. (system_impl.cpp:382)
[05:04:35|Debug] Discovered 1 component(s) (system_impl.cpp:514)
Discovered Autopilot from Client
[05:04:45|Warn ] Requesting Home Position Again. (telemetry_impl.cpp:245)
[05:04:45|Debug] Setting interval for msg id: 245 interval_ms:1000 (telemetry_server_impl.cpp:35)
[05:04:45|Debug] Request Param CAL_GYRO0_ID (mavlink_parameters.cpp:1544)
[05:04:45|Debug] Request Param CAL_ACC0_ID (mavlink_parameters.cpp:1544)
[05:04:45|Debug] Request Param CAL_MAG0_ID (mavlink_parameters.cpp:1544)
[05:04:45|Debug] Request Param SYS_HITL (mavlink_parameters.cpp:1544)
[05:04:46|Warn ] sending again after 0.507269 s, retries to do: 3  (512). (mavlink_command_sender.cpp:307)
[05:04:46|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:46|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:46|Warn ] sending again after 1.01468 s, retries to do: 2  (512). (mavlink_command_sender.cpp:307)
[05:04:46|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:47|Warn ] sending again after 1.52342 s, retries to do: 1  (512). (mavlink_command_sender.cpp:307)
[05:04:47|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:47|Warn ] Requesting Home Position Again. (telemetry_impl.cpp:245)
[05:04:47|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:47|Error] Retrying failed (512) (mavlink_command_sender.cpp:331)
[05:04:48|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:49|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:49|Warn ] Requesting Home Position Again. (telemetry_impl.cpp:245)
[05:04:50|Warn ] sending again after 0.510353 s, retries to do: 3  (512). (mavlink_command_sender.cpp:307)
[05:04:50|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:50|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:50|Warn ] sending again after 1.01855 s, retries to do: 2  (512). (mavlink_command_sender.cpp:307)
[05:04:50|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:51|Warn ] sending again after 1.52637 s, retries to do: 1  (512). (mavlink_command_sender.cpp:307)
[05:04:51|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:51|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:51|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:127)
[05:04:51|Warn ] Requesting Home Position Again. (telemetry_impl.cpp:245)
[05:04:51|Error] Retrying failed (512) (mavlink_command_sender.cpp:331)
[05:04:52|Warn ] sending again after 0.506235 s, retries to do: 3  (512). (mavlink_command_sender.cpp:307)
[05:04:52|Warn ] Request was for msg ID: 280 (mavlink_command_sender.cpp:313)
[05:04:52|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:52|Warn ] sending again after 1.01247 s, retries to do: 2  (512). (mavlink_command_sender.cpp:307)
[05:04:52|Warn ] Request was for msg ID: 280 (mavlink_command_sender.cpp:313)
[05:04:53|Warn ] sending again after 1.51953 s, retries to do: 1  (512). (mavlink_command_sender.cpp:307)
[05:04:53|Warn ] Request was for msg ID: 280 (mavlink_command_sender.cpp:313)
[05:04:53|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:53|Warn ] Requesting Home Position Again. (telemetry_impl.cpp:245)
[05:04:53|Error] Retrying failed (512) (mavlink_command_sender.cpp:331)
[05:04:54|Warn ] sending again after 0.508105 s, retries to do: 3  (512). (mavlink_command_sender.cpp:307)
[05:04:54|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:54|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:54|Warn ] sending again after 1.01385 s, retries to do: 2  (512). (mavlink_command_sender.cpp:307)
[05:04:54|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:55|Warn ] sending again after 1.52142 s, retries to do: 1  (512). (mavlink_command_sender.cpp:307)
[05:04:55|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:313)
[05:04:55|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
[05:04:55|Warn ] Requesting Home Position Again. (telemetry_impl.cpp:245)
[05:04:55|Debug] Request Param MY_PARAM (mavlink_parameters.cpp:1544)
[05:04:55|Debug] Missing Param MY_PARAM (mavlink_parameters.cpp:1555)
[... 156 lines cut ...]
[05:05:18|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
Vehicle is getting ready to arm
Altitude: 0 m
[05:05:19|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
Altitude: 0 m
[... cut ...]
[05:05:21|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
Altitude: 0 m
[05:05:21|Warn ] System status is usually fixed at 1 Hz (telemetry_impl.cpp:1939)
Vehicle is getting ready to arm
Altitude: 0 m
```
so the segfault is indeed gone. But the example has 2 issues:

1. the custom-param requested is case-sensitive so this is not working (just modify the case for `my_param`)
2. the vehicle is stuck at the check for `telemetry.health_all_ok() != true`

shall i close this issue here and create a new one for the adjustment of the autopilot_server example?

closed 2022-06-30 by julianoes