mavlink/MAVSDK #1842: Outdated code in takeoff_and_land.cpp example.

linuxguy123, 2022-07-05:
Details are here:
https://discuss.px4.io/t/error-get-system-id-was-not-declared-in-this-scope-auto-system-get-system-id-mavsdk/28162

Someone should at least update the example files with a note that they may no longer compile against the new API changes.   Even better would be build them in the test sequence to ensure they are always up to date.

JonasVautherin, 2022-07-05:
Would you mind describing the issue here, instead of pointing to another page?

This said, the examples are built by the CI, see [here](https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L106). Are you saying that something is wrong in the CI and it misses the `takeoff_and_land.cpp` example?

linuxguy123, 2022-07-05:
It must miss it.
I spent 3 hours trying to figure this out.  It's my first time using MAVSDK.
Line 75 has  `auto system = get_system(mavsdk);`
This causes a compile error `error: ‘get_system_id’ was not declared in this scope 289 | auto system = get_system_id(mavsdk);`
Apparently the new API method of doing this is `auto system = mavsdk.systems().at(0);`  per https://mavsdk.mavlink.io/main/en/cpp/api_changes.html

linuxguy123, 2022-07-06:
I got it wrong.  get_system was elsewhere in the file.

closed 2022-07-06 by JonasVautherin