mavlink/MAVSDK #1843: takeoff_and_landing.cpp won't build.

linuxguy123, 2022-07-06:
Copy and pasted takeoff_and_landing.cpp into test.cpp from https://github.com/mavlink/MAVSDK/blob/main/examples/takeoff_and_land/takeoff_and_land.cpp

CMakeLists.txt:
add_executable(test test.cpp)

# Libraries to link in 
target_link_libraries(test  
                MAVSDK::mavsdk
)

$ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/me/Development/myProject/build
[me@workstation1 build]$ make
[ 25%] Building CXX object src/CMakeFiles/test.dir/test.cpp.o
/home/me/Development/myProject/src/test.cpp: In function ‘std::shared_ptr<mavsdk::System> get_system(mavsdk::Mavsdk&)’:
/home/me/Development/myProject/src/test.cpp:37:13: error: ‘NewSystemHandle’ is not a member of ‘mavsdk::Mavsdk’
   37 |     Mavsdk::NewSystemHandle handle = mavsdk.subscribe_on_new_system([&mavsdk, &prom, &handle]() {
      |             ^~~~~~~~~~~~~~~
/home/me/Development/myProject/src/test.cpp:47:6: error: expected primary-expression before ‘)’ token
   47 |     });
      |      ^
make[2]: *** [src/CMakeFiles/test.dir/build.make:76: src/CMakeFiles/test.dir/test.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:115: src/CMakeFiles/test.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

It can't find NewSystemHandle. 

cd /usr/include/mavsdk
grep -r NewSystem
mavsdk.h:    using NewSystemCallback = std::function<void()>;
mavsdk.h:    void subscribe_on_new_system(const NewSystemCallback& callback);

linuxguy123, 2022-07-06:
It appears that NewSystemHandle has been replaced by NewSystemCallback.

It appears that the parameter list of subscribe_on_new_system has changed.

It appears that unsubscribe_on_new_system has been replaced by calling subscribe_on_new_system with null.  It also unsubscribes automatically. 

The get_system function in the examples needs to be updated.

JonasVautherin, 2022-07-06:
Which version of MAVSDK are you using with those examples? If you build MAVSDK from sources from branch `main`, then you can use the examples from branch `main`. If you use MAVSDK `v1.4.3`, then you should use the examples from tag `v1.4.3`.

linuxguy123, 2022-07-06:
I'll look into that.

linuxguy123, 2022-07-06:
OK, got it working.
The example link I was using pointed to version 1.1 ish code.

On Github, the example on the "main" branch still uses the old code.
https://github.com/mavlink/MAVSDK/blob/main/examples/takeoff_and_land/takeoff_and_land.cpp

However, if one selects version 1.4.3, it uses the new code.
https://github.com/mavlink/MAVSDK/blob/v1.4.3/examples/takeoff_and_land/takeoff_and_land.cpp

JonasVautherin, 2022-07-06:
> On GitHub, the example on the "main" branch still uses the old code.

I think there is some confusion here.

* The `main` branch is the development branch. It contains the latest everything (code, examples, etc).
* A release on GitHub is associated to a git [tag](https://git-scm.com/book/en/v2/Git-Basics-Tagging), which points the whole repo to a commit in the past.
* As a consequence, `main` keeps evolving, while tags (like `v1.4.3`) will never change.

I am pretty sure that the reason why the examples from `main` don't work when you build against the `v1.4.3` tag is because `main` is _newer_, not older. In fact, the next release coming from `main` will be `v2.0.0`, because (as you realised here) it is breaking the `v1.x.x` API.

Does that help?

linuxguy123, 2022-07-07:
Thanks for the explanation.

JonasVautherin, 2022-07-07:
Is there still a question, or should we close this issue?

closed 2022-07-09 by julianoes

violaBook, 2022-07-21:
> 
thanks a lot, it may really helps.

nello, 2022-10-16:
Not sure how this is meant to be "fixed"? Examples from `main` branch don't build against the current homebrew install. Is there a correct branch to be using that does build?

JonasVautherin, 2022-10-16:
What is the current homebrew install, and did you try to build the examples from the corresponding [release tag](https://github.com/mavlink/mavsdk/releases)?

As commented [above](https://github.com/mavlink/MAVSDK/issues/1843#issuecomment-1176818393), if you try to build an example from `main` (which is going towards v2) against an installation of MAVSDK v1, then probably it will fail (because moving from v1 to v2 means that there are breaking changes in the API).

Does that make sense?

nello, 2022-10-16:
Hi, I did manage to resolve this, so apologies.

Current `homebrew` is at v1.4.7, and the 1.4.7 tag for the repo worked for me. Sorry to bother you!

luketjones99, 2022-12-09:
Hi, I've gone through this discussion a few times and I'm still a little confused. (Apologies I'm completely new to PX4 and MAVSDK development). I cloned the most recent version of the repo and got everything setup properly, but I had the same issues when trying to run the takeoff and land example.

I was able to get it to build and work in simulation when I switched to the older v1.4.3 branch. But I was just wondering if there was something I should do so that I can run the examples with the newest code?

You mentioned that you have a CI system that should keep everything up to date, and I'm wondering if that is something I need to run manually to update the example files? (Apologies - I'm not familiar with how a CI system works). Just wanted to make sure I had everything up to date before I start practicing with my own examples. Thanks!

luketjones99, 2022-12-09:
To put it a bit more concisely: Is there a way to update the example files so that they will run with the most recent version? Or am I just misunderstanding your previous description?