mavlink/MAVSDK #1846: Command unsupported (511). (mavlink_command_sender.cpp:198)

linuxguy123, 2022-07-07:
test.cpp is [MAVSDK/takeoff_and_land.cpp at v1.4.3 · mavlink/MAVSDK · GitHub](https://github.com/mavlink/MAVSDK/blob/v1.4.3/examples/takeoff_and_land/takeoff_and_land.cpp)

The flight controller is a Pixhawk running Arduplane V3.10.0. Mission Planner 1.3.76 talks to it without issue.

The platform is Fedora 36, KDE, X11. Compiler is gcc.

When I run test.cpp (takeoff_and_land/takeoff_and_land.cpp), I get the following output:

./test serial:///dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00
[04:13:54|Info ] MAVSDK version: v1.4.3 (mavsdk_impl.cpp:20)
Waiting to discover system…
[04:13:54|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:482)
[04:13:54|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[04:13:54|Debug] Discovered 1 component(s) (system_impl.cpp:578)
Discovered autopilot
[04:13:55|Warn ] command unsupported (511). (mavlink_command_sender.cpp:198)
[04:13:55|Warn ] command unsupported (511). (mavlink_command_sender.cpp:198)
Setting rate failed: Unsupported

The code segment in question is:

` // Instantiate the telemetry plugin.
auto telemetry = Telemetry{system};
auto action = Action{system};

// We want to listen to the altitude of the drone at 1 Hz.
const auto set_rate_result = telemetry.set_rate_position(1.0);
if (set_rate_result != Telemetry::Result::Success) {
    std::cerr << "Setting rate failed: " << set_rate_result << '\n';
    return 1;
}`

How do I format a code block in this editor ?

JonasVautherin, 2022-07-07:
511 is https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL

Does Arduplace support it?

linuxguy123, 2022-07-08:
I just found this:

"Send a [SET_MESSAGE_INTERVAL](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL) command (within a [COMMAND_LONG](https://mavlink.io/en/messages/common.html#COMMAND_LONG) message) to precisely control the rate of an individual message. Note this is only supported on ArduPilot 4.0 and higher"

https://ardupilot.org/dev/docs/mavlink-requesting-data.html

I'll update Arduplane and report back.

I attached another Arduplane flight controller with newer firmware and it works.

JonasVautherin, 2022-07-08:
Awesome! Sounds like it's resolved then :blush:. Feel free to re-open if you feel like there is more to discuss!

closed 2022-07-08 by JonasVautherin

anjaliJT, 2023-01-27:
I'm using MAV_CMD_USER_4 for custom message but it's showing unsupported.
how to use this command without error.
i'm sending it like : 

vehicle.send_mavlink(vehicle.message_factory.command_long_encode(
            0, 0,  # target system, target component
            mavutil.mavlink.MAV_CMD_DO_SET_HOME,  # command
            0,  # confirmation
            0,  # param 1: 1 to use current position, 0 to use the entered values.
            0, 0, 0,  # params 2-4
            0.0, 0.0, 0.0))

@vehicle.on_message('*')
def message(n,m,m2):
    print(m2)

JonasVautherin, 2023-01-27:
Feels like this is unrelated to the question (which was for 511). Can you please open a new issue with a meaningful title? :see_no_evil:

cryptik, 2023-09-06:
This is a closed issue, but I am going to add some latest behavior in case this may need to be reopened.  I am using mavsdk_server from version MAVSDK_Python that I built from source using git tag 1.4.8 on Debian ARM64.  The mavsdk_server binary runs just fine and reports v1.4.16 on startup.

The problem I have is that if I make the following call:
`await self._drone.core.set_mavlink_timeout(1.0)`

Any subsequent calls to 
```
await self._drone.telemetry.set_rate_gps_info(3.0)
await self._drone.telemetry.set_rate_position_velocity_ned(10.0)
await self._drone.telemetry.set_rate_home(5.0)
await self._drone.telemetry.set_rate_battery(1.0)
```
will result in the error 
```
[05:27:56|Warn ] Received ack for not-existing command: 511! Ignoring... (mavlink_command_sender.cpp:258)
```
If I remove the call to `set_mavlink_timeout` completely, the calls to `set_rate_` for the other message streams works just fine.

I am running the latest Ardu firmware on CubePilot.  I know that `mavsdk_server` primarily supports PX4, so I was not sure if this behavior is by design.