mavlink/MAVSDK #1860: What happens when communication is disconnected

valenbase, 2022-07-15:
Using mavsdk as an autopilot server, what would happen when communication is disconnected?

Where can I get the exception handling description?

THX.

JonasVautherin, 2022-07-15:
Would this do it? https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/core/core.proto#L13

julianoes, 2022-07-15:
I don't know, it's a good question. What are you trying to do or solve?

paulholmstedt, 2022-08-03:
I have a similar issue: When using a Cube Orange connected via USB serial communication and the USB cable is plugged out my MAVSDK based software is unable to reconnect when the USB cable is re-inserted. Is it possible to get access to the Connection object and perform a Stop/Start call? I use the add_any_connection method but there is no remove_any_connection in case the communication dies. Or am I missing something?

julianoes, 2022-08-03:
@paulholmstedt what you could do is to destruct `Mavsdk` and create it again. `Mavsdk` would take the connections down with itself.

However, I agree, that an `remove_any_connection` might be nice.

paulholmstedt, 2022-08-05:
Would it possible to add a method (get_any_connection) in mavsdk api which allow you to get a reference to the connection instance itself i.e. 

std::shared_ptr<Connection>& connection = mavsdk.get_any_connection(const std::string& connection_url);

Then the programmer can call connection.stop(); and connection.start(); to deconstruct and construct the connection again without deleting all instances of classes already created?

julianoes, 2022-08-05:
Yer that would be a good option. It would be like the new subscription handle.

julianoes, 2025-12-16:
@Ryanf55 so you would want the equivalent of [Mavsdk::subscribe_on_new_system](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_mavsdk.html#classmavsdk_1_1_mavsdk_1a5b7c958ad2e4529dc7b950ab26618575) but `subscribe_on_lost_system`?

What about using [System::subscribe_is_connected](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_system.html#classmavsdk_1_1_system_1aae68747c23976fa7eb63ec0762493263) instead?

Ryanf55, 2025-12-16:
> [@Ryanf55](https://github.com/Ryanf55) so you would want the equivalent of [...]

That works!